platformio/platformio-atom-ide #1950: InstallationManager

popoff1998, 2018-06-26:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.28.0
PIO IDE: v2.1.7
System: Windows_NT, 6.1.7601, x64

# Exception
```
PIP: Command "python setup.py egg_info" failed with error code 1 in c:\users\tonin.uco\.platformio\.cache\tmp-19688tok9faxsbhip\pip-install-z_b0ou\semantic-version\

```

ivankravets, 2018-06-26:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

Henkeh, 2018-07-15:
I am experiencing similar problems. Also there are several other reports issue list that are also similar.

My error report:

# Configuration

VSCode: 1.25.1
PIO IDE: v0.17.2
System: Windows_NT, 10.0.17134, x64

# Exception
```
PIP: Command "python setup.py egg_info" failed with error code 1 in c:\users\ramon\.platformio\.cache\tmp-4456crsqbaplmbpo\pip-req-build-yq8c1s\

```

As for the commands provide above; I made sure all other Python versions where removed, so none of these commands give useful output I guess.

Any help would be appreciated!

Henkeh, 2018-07-15:
I fiddled around a bit more and found a solution for my situation. I am running Comodo Internet Security which has an auto-containment feature. This interfered with the installation. Everything installed just fine after disabling this feature for a couple of minutes.

Hope something similar works for other people!

kartoffelfighter, 2018-07-15:
Found a solution, platformio core was blocked by windows defender.
I just disabled in the security center at app and browsercontrol the feature "check files and apps"
Maybe the exact translation in win10 english is different, i am using a german version

maybe this helps somebody

ivankravets, 2018-07-17:
Please try this solution http://docs.platformio.org/en/latest/installation.html#virtual-environment

commit 10eec9c9 (2018-07-19): Fix "PIP: Command "python setup.py egg_info" failed" // Resolve #1950
closed 2018-07-17 by ivankravets