platformio/platformio-atom-ide #1954: InstallationManager

dpondaco, 2018-06-26:
# Description of problem
Leave a comment..error that occurs on atom/PIO startup from develop version of ide .Install Virtualenv globally: The system cannot find the path specified. Error: spawn C:\Users\mickey_tablet.platformio\penv\Scripts\python.exe ENOENT

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.28.0
PIO IDE: v2.1.7
System: Windows_NT, 10.0.17134, ia32

# Exception
```
Install Virtualenv globally: The system cannot find the path specified.
Error: spawn C:\Users\mickey_tablet\.platformio\penv\Scripts\python.exe ENOENT
```

Henkeh, 2018-07-15:
I have seen this error as well during my struggles to get PIO installed. Possible solution can be found in my post in:
https://github.com/platformio/platformio-atom-ide/issues/1950

ivankravets, 2018-07-17:
Please try this solution http://docs.platformio.org/en/latest/installation.html#virtual-environment

closed 2018-07-17 by ivankravets