platformio/platformio-atom-ide #1960: platformio-ide breaks pyls (python language server)

HetorusNL, 2018-06-28:
before installing platformio-ide and when disabling the platformio-ide package, the python language server (pyls) was working correctly.
after installing platformio-ide, Atom complains about:
Unable to start the Python language server.
Make sure to install pyls 0.19 or newer by running:
python -m pip install 'python-language-server[all]'

Python: 2.7 (running in virtual-env)
OS: Kali Linux (based on Debian Stretch)
Atom: 1.27.2 (started from console in python virtual-env)

[image]
[image]

martenjacobs, 2018-12-10:
I have the same issue under macOS 10.14 (18A391)

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets