platformio/platformio-atom-ide #1971: Error: Invalid `platformio.ini`, project configuration file: 'No option 'upload_port' in section: 'common''

AlexTakt, 2018-07-02:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.28.0
PIO IDE: v2.1.7
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Invalid `platformio.ini`, project configuration file: 'No option 'upload_port' in section: 'common''

```
According to this:
http://docs.platformio.org/en/latest/projectconf/section_env_upload.html#upload-port
If upload_port isn’t specified, then PlatformIO will try to detect it automatically.

Not a big deal, but annoying

ivankravets, 2018-07-02:
You have broken PlatformIO.ini file

closed 2018-07-02 by ivankravets

AlexTakt, 2018-07-03:
Hi, can you be a bit more specific, or point to more specific answer please?
Thanks

ivankravets, 2018-07-03:
Please share your `platformio.ini`

AlexTakt, 2018-07-03:
Thanks for prompt response.
This one is working, but requires explicit port number.

ivankravets, 2018-07-10:
Check sections and template variables

ivankravets, 2018-07-10:
http://docs.platformio.org/en/latest/projectconf/dynamic_variables.html

AlexTakt, 2018-07-10:
Thanks for that. :)

AntonioBittencourt, 2020-03-12:
I`m getting the same error and in my platformio.ini appears:
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp12e]
platform = espressif8266
board = esp12e
framework = arduino

 should i change for what appears in the link(https://docs.platformio.org/en/latest/projectconf/dynamic_variables.html):

ivankravets, 2020-03-12:
Please forward to https://community.platformio.org/