platformio/platformio-atom-ide #1973: InstallationManager

ricain59, 2018-07-02:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.28.0
PIO IDE: v2.1.7
System: Windows_NT, 6.1.7601, x64

# Exception
```
PIP: Command "python setup.py egg_info" failed with error code 1 in c:\users\ricain\.platformio\.cache\tmp-9816gcbsg1degk3f\pip-req-build-3loxos\

```

Henkeh, 2018-07-15:
I have seen this error as well during my struggles to get PIO installed. Possible solution can be found in my post in:
https://github.com/platformio/platformio-atom-ide/issues/1950

ivankravets, 2018-07-19:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1950

closed 2018-07-19 by ivankravets