platformio/platformio-atom-ide #1979: InstallationManager

neiseng, 2018-07-05:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.28.0
PIO IDE: v2.1.7
System: Windows_NT, 6.1.7601, x64

# Exception
```
Install Virtualenv globally: ϵͳ�Ҳ���ָ����·����
Error: spawn C:\Users\Administrator\.platformio\penv\Scripts\python.exe ENOENT
```

Henkeh, 2018-07-15:
I have seen this error as well during my struggles to get PIO installed. Possible solution can be found in my post in:
https://github.com/platformio/platformio-atom-ide/issues/1950

ivankravets, 2018-07-17:
Please try this solution http://docs.platformio.org/en/latest/installation.html#virtual-environment

closed 2018-07-17 by ivankravets