platformio/platformio-atom-ide #1981: InstallationManager

rinath, 2018-07-05:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.27.2
PIO IDE: v2.1.7
System: Linux, 4.15.0-24-generic, x64

# Exception
```
Install Virtualenv globally: /usr/local/bin/python2.7: No module named pip

```

auralmenace, 2018-07-05:
I searched for this issue elsewhere, but this is all I found.

I have the same issue. I checked the following in the PlatformIO terminal:
    $ pip -V
    $ pip 9.0.1 from /usr/lib/python2.7/dist-packages (python 2.7)

I copied the pip module folder to /usr/lib/bin - no change.

the installer for PlatformIO seems to work OK, but gets stuck on the last item and then the pip error appears.

auralmenace, 2018-07-06:
Sorry, forgot the details:
I'm running Ubuntu 18.10

ivankravets, 2018-07-08:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

auralmenace, 2018-07-13:
@ivankravets - sorry for the delay

$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

echo $PYTHONPATH

$ python --version
Python 2.7.15

$ pip --version
pip 9.0.1 from /usr/lib/python2.7/dist-packages (python 2.7)

$ pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
asn1crypto (0.24.0)
configparser (3.5.0)
cryptography (2.2.2)
entrypoints (0.2.3.post1)
enum34 (1.1.6)
idna (2.6)
ipaddress (1.0.17)
keyring (13.1.0)
keyrings.alt (3.0)
pip (9.0.1)
pycrypto (2.6.1)
pygobject (3.28.2)
pyxdg (0.25)
SecretStorage (2.3.1)
setuptools (39.2.0)
six (1.11.0)
wheel (0.30.0)

auralmenace, 2018-07-13:
@ivankravets - Thanks for the assistance.

$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

$ echo $PYTHONPATH

$ python --version
Python 2.7.15

$ pip --version
pip 9.0.1 from /usr/lib/python2.7/dist-packages (python 2.7)

pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
asn1crypto (0.24.0)
configparser (3.5.0)
cryptography (2.2.2)
entrypoints (0.2.3.post1)
enum34 (1.1.6)
idna (2.6)
ipaddress (1.0.17)
keyring (13.1.0)
keyrings.alt (3.0)
pip (9.0.1)
pycrypto (2.6.1)
pygobject (3.28.2)
pyxdg (0.25)
SecretStorage (2.3.1)
setuptools (39.2.0)
six (1.11.0)
wheel (0.30.0)

ivankravets, 2018-07-15:
Please fix that manually http://docs.platformio.org/en/latest/installation.html#virtual-environment

closed 2018-07-15 by ivankravets

auralmenace, 2018-07-16:
@ivankravets - Thanks for the help:
$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

$ echo $PYTHONPATH

$ python --version
Python 2.7.15

pip --version
pip 9.0.1 from /usr/lib/python2.7/dist-packages (python 2.7)

pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
asn1crypto (0.24.0)
configparser (3.5.0)
cryptography (2.2.2)
entrypoints (0.2.3.post1)
enum34 (1.1.6)
idna (2.6)
ipaddress (1.0.17)
keyring (13.1.0)
keyrings.alt (3.0)
pip (9.0.1)
pycrypto (2.6.1)
pygobject (3.28.2)
pyxdg (0.25)
SecretStorage (2.3.1)
setuptools (39.2.0)
six (1.11.0)
wheel (0.30.0)

auralmenace, 2018-07-17:
echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

auralmenace, 2018-07-19:
I am attempting a penv setup. My entire Ubuntu installation got screwed up after trying to fix all of the multiple Python installations. I just did a clean install of Atom and performing a virtual environment by hand _not_ with the Atom package installer.

auralmenace, 2018-07-19:
@ivankravets 
After removing the ~/.platformio folder, creating an independent environment, and doing a manual installation into that environment, and _then_ installing the ATOM platformIO package, PlatformIO is running fine - but given that I wiped my system, I can't be sure where the fix occurred. I still have to fix the PATH to point to the virtualenv path. 
So, I'm sorry, I don't believe that I provided you with a solid fix that can be handed off to others with the same issue.
Thanks for the help! PlatformIO has been indispensable to me because it keep everything in ATOM and I don't have to jump around in environments.
Thanks to you and all of the team!