platformio/platformio-atom-ide #1994: InstallationManager

ethan92429, 2018-07-09:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.28.1
PIO IDE: v2.1.7
System: Darwin, 17.5.0, x64

# Exception
```
PIP: Exception:
Traceback (most recent call last):
  File "/Users/ethankeller/.platformio/penv/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/Users/ethankeller/.platformio/penv/lib/python2.7/site-packages/pip/commands/install.py", line 342, in run
    prefix=options.prefix_path,
  File "/Users/ethankeller/.platformio/penv/lib/python2.7/site-packages/pip/req/req_set.py", line 784, in install
    **kwargs
  File "/Users/ethankeller/.platformio/penv/lib/python2.7/site-packages/pip/req/req_install.py", line 851, in install
    self.move_wheel_files(self.source_dir, root=root, prefix=prefix)
  File "/Users/ethankeller/.platformio/penv/lib/python2.7/site-packages/pip/req/req_install.py", line 1064, in move_wheel_files
    isolated=self.isolated,
  File "/Users/ethankeller/.platformio/penv/lib/python2.7/site-packages/pip/wheel.py", line 352, in move_wheel_files
    clobber(source, lib_dir, True)
  File "/Users/ethankeller/.platformio/penv/lib/python2.7/site-packages/pip/wheel.py", line 330, in clobber
    shutil.copyfile(srcfile, destfile)
  File "/Users/ethankeller/.platformio/penv/lib/python2.7/shutil.py", line 97, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied: '/lib/python2.7/site-packages/bottle.pyc'
You are using pip version 9.0.3, however version 10.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

```

ivankravets, 2018-07-17:
Please install PlatformIO Core manually http://docs.platformio.org/en/latest/installation.html#virtual-environment

closed 2018-07-17 by ivankravets

ethan92429, 2018-07-17:
Why do I have to install manually? Why can't it install automatically?

ivankravets, 2018-07-17:
It should work automatically. Could you remove `/Users/ethankeller/.platformio/penv` folder and restart IDE?

ethan92429, 2018-07-18:
Hmm, I get the same permission error as posted before... Any other suggestion?

ivankravets, 2018-07-18:
Could you try to upgrade system `pip`  using system terminal via
```
sudo pip install -U pip
```