platformio/platformio-atom-ide #2004: InstallationManager Python not found

ChristianKrausse, 2018-07-11:
# Description of problem
It ist true, there is no directory at %USERPROFILE%\.platformio\ like penv 
there is a python located at %USERPROFILE%\.platformio\python27\python.exe should this be ignored or anyhow fixed?

# Configuration
Atom: 1.28.1
PIO IDE: v2.1.7
System: Windows_NT, 10.0.17692, x64

# Exception
```
Install Virtualenv globally: Das System kann den angegebenen Pfad nicht finden.
Error: spawn C:\Users\MyUserName\.platformio\penv\Scripts\python.exe ENOENT
```

Henkeh, 2018-07-15:
I have seen this error as well during my struggles to get PIO installed. Possible solution can be found in my post in:
https://github.com/platformio/platformio-atom-ide/issues/1950

kartoffelfighter, 2018-07-15:
Had the same problem, solution here:
https://github.com/platformio/platformio-atom-ide/issues/1950#issuecomment-405094459

ivankravets, 2018-07-17:
Please try this solution http://docs.platformio.org/en/latest/installation.html#virtual-environment

closed 2018-07-17 by ivankravets