platformio/platformio-atom-ide #2006: Start PIO Home Server

hugoru, 2018-07-11:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.28.1
PIO IDE: v2.1.7
System: Windows_NT, 10.0.17666, x64

# Exception
```
Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 24, in init cmd_debug
  File "C:\Users\garri\.platformio\packages\contrib-pysite\twisted\internet\protocol.py", line 17, in <module>
    from twisted.python import log, failure, components
  File "C:\Users\garri\.platformio\packages\contrib-pysite\twisted\python\log.py", line 26, in <module>
    from twisted.logger import (
  File "C:\Users\garri\.platformio\packages\contrib-pysite\twisted\logger\__init__.py", line 89, in <module>
    from ._levels import InvalidLogLevelError, LogLevel
  File "C:\Users\garri\.platformio\packages\contrib-pysite\twisted\logger\_levels.py", line 9, in <module>
    from constantly import NamedConstant, Names
ImportError: No module named constantly

```

wegunterjr, 2018-07-12:
I ama getting the same problem. Is the server down?

wegunterjr, 2018-07-12:
I did
`platformio update` from the PlatformIO terminal in Atom, and 
 `contrib-pysite                  @ 0.3.0          [0.3.1]`
noticed that the contrib-pysite was not up to date.  When it updated, i was able to launch the PlatformIO.

ivankravets, 2018-07-17:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/165

closed 2018-07-17 by ivankravets