platformio/platformio-atom-ide #2010: Error: Could not install 'toolchain-atmelavr'

chonuchan, 2018-07-13:
### Configuration

**Operating system**:  ubuntu 16.04 64 bit

**PlatformIO Version** (`platformio --version`): 3.6.0a6

### Description of problem

Installed platformio-ide through atom 1.26.1, 

#### Steps to Reproduce

1.Install platformio-ide through atom 1.26.1
2. Create a new project, select any board, return **Error 1.
3. Followed http://bit.ly/faq-package-manager Method 3, at system terminal:
3a. adding ~/.platformio/penv/bin/platformio  to Bash command, reboot.
3b. run platformio home at project folder, a platformio homepage pop up at browser
3c. run platformio run --target upload at project folder, return **Error2.

### Actual Results

**Error 1:************

Could not initialize project
PIO Core Call Error: The next files/directories have been created in /home/xxx/Documents/PlatformIO/Projects/bbb platformio.ini - Project Configuration File src - Put your source files here lib - Put here project specific (private) libraries Error: Could not install 'toolchain-atmelavr' with version requirements '~1.40902.0' for your system 'linux_x86_64'. Please try this solution -> http://bit.ly/faq-package-manager
************************

**Error 2:*************
Processing teensy20 (platform: teensy; board: teensy20; framework: arduino)
--------------------------------------------------------------------------------
PackageManager: Installing toolchain-atmelavr @ ~1.40902.0
Downloading  [####################################]  100%          
Unpacking  [------------------------------------]    0%
Warning! Package Mirror: Could not extract `avr/bin/ar` to `/home/hunter/.platformio/packages/_tmp_installing-RIlujo-package`. Try to disable antivirus tool or check this solution -> http://bit.ly/faq-package-manager
Looking for another mirror...
Downloading...
Unpacking  [------------------------------------]    0%
Warning! Package Mirror: Could not extract `avr/bin/ar` to `/home/hunter/.platformio/packages/_tmp_installing-RO30EL-package`. Try to disable antivirus tool or check this solution -> http://bit.ly/faq-package-manager
Looking for another mirror...
Error: Could not install 'toolchain-atmelavr' with version requirements '~1.40902.0' for your system 'linux_x86_64'.

Please try this solution -> http://bit.ly/faq-package-manager

***************************

### Expected Results
installing toolchain-atmelavr to ~/.platformio/packages/

### Additional info

** NO Antivirus installed

**Running in Bash: ls -l ~/.platformio , return:

-rw-rw-r-- 1 xxx xxx  188 Jul 13 02:18 appstate.json
-rw-rw-r-- 1 xxx xxx  483 Jul 13 02:23 homestate.json
drwxrwxr-x 2 xxx xxx 4096 Jul 13 02:18 lib
drwxrwxr-x 5 xxx xxx 4096 Jul 13 02:24 packages
drwxrwxr-x 6 xxx xxx 4096 Jul 13 02:18 penv
drwxrwxr-x 3 xxx xxx 4096 Jul 13 02:18 platforms

chonuchan, 2018-07-13:
I am trying to bypass this issue by using Method 4 in  http://bit.ly/faq-package-manager.

It seems platformio manage to install most of the packages, except for these two:
tool-avrdude
toolchain-atmelavr

After manually installing them, I can now upload to a board. But still, I think there is a permission issue with some packages...

chonuchan, 2018-07-13:
Interestingly, I notice from the Linux file manager that a new folder is transient created at ~/.platformio/packages/ for the failed installation. Might be it is now a permission issue, but instead an unzipping issue? **What package does platformio use to unzip**?

ivankravets, 2018-07-13:
Fixed in https://github.com/platformio/platformio-core/issues/1733

closed 2018-07-13 by ivankravets

chonuchan, 2018-07-13:
Thanks, Ivan