platformio/platformio-atom-ide #2014: InstallationManager

angelmicelti, 2018-07-14:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.28.2
PIO IDE: v2.1.7
System: Windows_NT, 10.0.17134, x64

# Exception
```
Install Virtualenv globally: El sistema no puede encontrar la ruta especificada.
Error: spawn C:\.platformio\penv\Scripts\python.exe ENOENT
```

Henkeh, 2018-07-15:
I have seen this error as well during my struggles to get PIO installed. Possible solution can be found in my post in:
https://github.com/platformio/platformio-atom-ide/issues/1950

ivankravets, 2018-07-17:
Please try this solution http://docs.platformio.org/en/latest/installation.html#virtual-environment

closed 2018-07-17 by ivankravets