platformio/platformio-atom-ide #2026: Latest Atom Update Breaks PlatformIO IDE. Cannot find module 'tslib'.

solinvictus21, 2018-07-19:
After updating to Atom 1.28.2, the PlatformIO IDE plugin for Atom is now broken with the error "Cannot find module 'tslib'." Already tried deleting ~/.platformio and ~/.atom and restarting the Atom IDE and that did not correct the issue. This is on MacOS. Unsure if other platforms are affected.

Full stack trace is below...

Error: Cannot find module 'tslib'
    at Module._resolveFilename (module.js:543:15)
    at Module._resolveFilename (/Applications/Atom.app/Contents/Resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.get_Module._resolveFilename (/Applications/Atom.app/Contents/Resources/app/src/module-cache.js:354:58)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app.asar/static/index.js:40:43)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at Object.<anonymous> (/Users/kdavis4/.atom/packages/platformio-ide/node_modules/jsonrpc-lite/jsonrpc.js:7:15)
    at Object.<anonymous> (/Users/kdavis4/.atom/packages/platformio-ide/node_modules/jsonrpc-lite/jsonrpc.js:374:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app.asar/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at t (/Users/kdavis4/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1:355)
    at Object.<anonymous> (/Users/kdavis4/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1:1173)
    at Object.<anonymous> (/Users/kdavis4/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app.asar/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at Object.<anonymous> (/Users/kdavis4/.atom/packages/platformio-ide/lib/maintenance.js:24:30)
    at Object.<anonymous> (/Users/kdavis4/.atom/packages/platformio-ide/lib/maintenance.js:122:2)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)

Ulli2k, 2018-07-19:
same issue.
Is there a way to downgrade? the plattformio-ide?

yoyo23, 2018-07-19:
Also same here. OS X 10.12

ivankravets, 2018-07-19:
Sorry for the issue, we have just released 2.1.11 and fixed this problem.

See related issue https://github.com/teambition/jsonrpc-lite/issues/4

closed 2018-07-19 by ivankravets

yoyo23, 2018-07-19:
Thank you very much. 
Problem fixed.

solinvictus21, 2018-07-20:
Confirmed fixed for me as well.