platformio/platformio-atom-ide #2039: C/C++ project index rebuild failed

kd6hq, 2018-07-24:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.28.2
PIO IDE: v2.1.11
System: Windows_NT, 6.2.9200, x64

# Exception
```
Error: 
Error: Invalid environment 'common'. The name must start with 'env:'

```

kd6hq, 2018-07-24:
[platformio.txt](https://github.com/platformio/platformio-atom-ide/files/2223498/platformio.txt)
any thoughts?

ivankravets, 2018-07-24:
Add 2 spaces before https://github.com/trinamic/TMC26XStepper.git

kd6hq, 2018-07-25:
Thank you for responding.
I tried the 2 spaces but no luck.
So I did a search (project) for anywhere "TMC26XStepper.git" is used.
Only 2 places, one is part of a commit, and the other is in "platformio.ini".
Platformio.ini now looks like this attached file.
I also note that I get the error message even before I try to compile.  Several seconds after I open up the Project.
[platformio.txt](https://github.com/platformio/platformio-atom-ide/files/2225973/platformio.txt)

ivankravets, 2018-07-26:
It seems that you use different projects. Please check project paths.

closed 2018-07-26 by ivankravets