platformio/platformio-atom-ide #2042: C/C++ project index rebuild failed

liljuanton, 2018-07-27:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.28.2
PIO IDE: v2.1.11
System: Darwin, 17.2.0, x64

# Exception
```
Error: Traceback (most recent call last):
  File "/Users/liljuanton/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/Users/liljuanton/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Users/liljuanton/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Users/liljuanton/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/liljuanton/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/liljuanton/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/liljuanton/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/liljuanton/.platformio/penv/lib/python2.7/site-packages/platformio/commands/init.py", line 102, in cli
    pg = ProjectGenerator(project_dir, ide, env_name)
  File "/Users/liljuanton/.platformio/penv/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 35, in __init__
    self._gather_tplvars()
  File "/Users/liljuanton/.platformio/penv/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 151, in _gather_tplvars
    self._tplvars.update(self.get_project_build_data())
  File "/Users/liljuanton/.platformio/penv/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 75, in get_project_build_data
    raise result.exception
SSLError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /MarlinFirmware/U8glib-HAL/archive/dev.zip (Caused by SSLError(SSLError(1, u'[SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)'),))

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

```

alexge233, 2018-08-05:
+1 I'm also seeing this. All I did was install platformIO and open the Marlin folder.

ivankravets, 2018-08-30:
Fixed in the latest version of Atom and PlatformIO IDE package. Please update.

closed 2018-08-30 by ivankravets

workshopninja, 2019-05-30:
Latest Marlin 2.0 release - attempting to auto build for re-arm board 

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip; board: nxp_lpc1768; framework: arduino)
--------------------------------------------------------------------------------------------------------------------
LibraryManager: Installing U8glib-HAL
Error: Please read http://bit.ly/package-manager-ioerror
Error: Traceback (most recent call last):
  File "/Users/workshopninja/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/Users/workshopninja/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Users/workshopninja/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Users/workshopninja/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/workshopninja/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/workshopninja/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/workshopninja/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/workshopninja/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 107, in cli
    result = (envname, ep.process())
  File "/Users/workshopninja/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 207, in process
    result = self._run()
  File "/Users/workshopninja/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 302, in _run
    self.verbose)
  File "/Users/workshopninja/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 325, in _autoinstall_libdeps
    ctx.invoke(cmd_lib_install, libraries=[lib], silent=not verbose)
  File "/Users/workshopninja/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/workshopninja/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "/Users/workshopninja/.platformio/penv/lib/python2.7/site-packages/platformio/commands/lib.py", line 103, in lib_install
    library, silent=silent, interactive=interactive, force=force)
  File "/Users/workshopninja/.platformio/penv/lib/python2.7/site-packages/platformio/managers/lib.py", line 352, in install
    force=force)
  File "/Users/workshopninja/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 721, in install
    name, url, requirements, track=True)
  File "/Users/workshopninja/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 508, in _install_from_url
    dlpath = self.download(url, tmp_dir, sha1)
  File "/Users/workshopninja/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 199, in download
    raise e
SSLError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /MarlinFirmware/U8glib-HAL/archive/dev.zip (Caused by SSLError(SSLError(1, u'[SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)'),))