platformio/platformio-atom-ide #2059: Installe rhangs at atom dependancies step

stibinator, 2018-08-21:
I just tried installing Platformio IDE through the GUI package manager in Atom. It seemeed to be successful, but none of the menus or commands were visible, even after a restart. So when I opened the Platformio settings this popped up, and eventually got to the state it is in the picture, where it has hung ever since

[image]

stibinator, 2018-08-21:
I went through the config file and found all the dependancies, and then used apm to install them manually. looks like the one that is holding up the parade is Build.

ndi123, 2018-08-21:
Got the same issue. @stibinator Which dependencies did you install manually to make it work?

stibinator, 2018-08-22:
In the config.js file in the platformio there's a line `export const ATOM_DEPENDENCIES =...` I just copied every dependency listed there and installed with `apm install`. They all worked except for `build`. The list I got was: 
`'busy-signal' 'platformio-ide-terminal' 'platformio-ide-debugger' 'autocomplete-clang' 'linter-ui-default' 'intentions' 'linter' 'linter-gcc' 'language-ini' 'tool-bar' 'file-icons' 'build-platformio' 'build'`

roundy9, 2018-08-22:
I'm facing the same problem.  Were you able to find a way to manually install 'build' so there wouldn't be a hangup on the dependencies?

mienaikoe, 2018-08-22:
Having this problem too. Manually installing 'build' silently hangs for me as well.

This bug actually prevents Atom from doing anything else because the installation modal is overlaid on the whole main window, so it's pretty alarming.

ivankravets, 2018-08-22:
Related issue https://github.com/noseglid/atom-build/issues/569

@noseglid do you have any ideas?

KayIO, 2018-08-23:
I have the samissue on two different windows 10 computer. So I think it's a bigger problem. It seems that it's working with the node.js library.
[image]
Even when you stop Atom... the node.js task is running without any reasons and results.
I hink it's pretty annoying not to see, what steps the installer is doing. So it makes it really hard to locate the problem.

F1LYS, 2018-08-23:
same issue on W10 x64, seems to hang on getmac -> editions dependencies install...
apm hangs at the same point...

stibinator, 2018-08-24:
I'd say this is the culprit:
https://github.com/noseglid/atom-build/issues/569#issuecomment-415131142

DeltaZero, 2018-09-09:
Solved this problem by going to CMD and running `apm install build` while Atom is open and PlatformIO is stuck on Atom Dependencies!

ivankravets, 2018-11-29:
It seems that issue was fixed. Please reopen if you still experience it.

closed 2018-11-29 by ivankravets

0dUb0, 2019-06-26:
It seems that I have the same issue, stuck on dependencies installation. 
I can't install platformio terminal direclty :

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-pty-prebuilt@0.7.6 install: `prebuild-install || node scripts/install.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-pty-prebuilt@0.7.6 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

kulfuerst, 2019-08-14:
So what is the solution? Could someone please post information on how this can be fixed? I can't get past the Atom dependencies either. This installation is a nightmare

ivankravets, 2019-08-14:
Atom is a deprecated editor. Please use PlatformIo IDE for VSCode => https://platformio.org/platformio-ide

CodilX, 2019-08-19:
Couldn't get passed the infinite "Atom dependencies" screen. What finally worked:

1. Uninstall Atom through Add or Remove Programs
2. Remove the following directories (btw, great uninstaller that leaves almost all files on the drive 👍):

   - C:/ProgramData/USERNAME/atom <- this one was crucial, the directory is not present with 1.38.2 but is created with 1.40 and that creates an error when installing platformio-ide-terminal
   - C:/Users/USERNAME/.atom
   - C:/Users/USERNAME/.platformio
   - C:/Users/USERNAME/AppData/Local/atom
   - C:/Users/USERNAME/AppData/Roaming/atom
3. Download and install [Atom1.38.2](https://github.com/atom/atom/releases/tag/v1.38.2) 
4. Quickly go into settings (`CTRL+ ,`), disable auto updates and then close Atom
5. Open up the command line and install the following packages

```
apm install platformio-ide@2.3.1
apm install platformio-ide-debugger@1.2.6
apm install platformio-ide-terminal@2.9.1
```
Then you can open Atom and be greated with that lovely installing platformio screen, but this time it should finally finish.

Not sure if `platformio-ide-debugger` is needed at this stage. `platformio-ide` is installed without any errors in terminal or through Atom, but `platformio-ide-terminal` is what was not installing before removing all the directories and still wouldn't install through GUI on 1.38.2

[image]

brianscottwilson, 2019-12-11:
I'm getting what look like the same issue (indefinite installation issue).  When I tried the "apm install ..." of the dependencies, I got the following.  It appears there is a self signed certificate somewhere in the  process that is raising issues.  Any chance someone will take a look at this?

```
$ apm install platformio-ide-terminal
Installing platformio-ide-terminal to /Users/bxw85a3/.atom/packages ✗

> node-pty-prebuilt-multiarch@0.9.0-beta21.legacy install /private/var/folders/1l/41j8fsys3ds6f5356d08fv5h0000gn/T/apm-install-dir-1191111-19511-1lat50k.cidpj/node_modules/platformio-ide-terminal/node_modules/node-pty-prebuilt-multiarch
> prebuild-install || node scripts/install.js

prebuild-install WARN install self signed certificate in certificate chain
gyp WARN install got an error, rolling back install
gyp ERR! configure error
gyp ERR! stack Error: self signed certificate in certificate chain
gyp ERR! stack     at TLSSocket.onConnectSecure (_tls_wrap.js:1055:34)
gyp ERR! stack     at TLSSocket.emit (events.js:182:13)
gyp ERR! stack     at TLSSocket._finishInit (_tls_wrap.js:635:8)
gyp ERR! System Darwin 18.7.0
gyp ERR! command "/private/var/folders/1l/41j8fsys3ds6f5356d08fv5h0000gn/T/AppTranslocation/355F5BE1-C904-4684-BA03-3A16D18D2E49/d/Atom.app/Contents/Resources/app/apm/bin/node" "/private/var/folders/1l/41j8fsys3ds6f5356d08fv5h0000gn/T/AppTranslocation/355F5BE1-C904-4684-BA03-3A16D18D2E49/d/Atom.app/Contents/Resources/app/apm/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /private/var/folders/1l/41j8fsys3ds6f5356d08fv5h0000gn/T/apm-install-dir-1191111-19511-1lat50k.cidpj/node_modules/platformio-ide-terminal/node_modules/node-pty-prebuilt-multiarch
gyp ERR! node -v v10.2.1
gyp ERR! node-gyp -v v3.7.0
gyp ERR! not ok
npm WARN enoent ENOENT: no such file or directory, open '/private/var/folders/1l/41j8fsys3ds6f5356d08fv5h0000gn/T/apm-install-dir-1191111-19511-1lat50k.cidpj/package.json'
npm WARN apm-install-dir-1191111-19511-1lat50k.cidpj No description
npm WARN apm-install-dir-1191111-19511-1lat50k.cidpj No repository field.
npm WARN apm-install-dir-1191111-19511-1lat50k.cidpj No README data
npm WARN apm-install-dir-1191111-19511-1lat50k.cidpj No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-pty-prebuilt-multiarch@0.9.0-beta21.legacy install: `prebuild-install || node scripts/install.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-pty-prebuilt-multiarch@0.9.0-beta21.legacy install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/bxw85a3/.atom/.apm/_logs/2019-12-11T15_20_50_185Z-debug.log
```

Emilio-IT, 2020-01-20:
Thanks to @stibinator , I managed to solve it on Mac OS 10.15.2 Catalina with Atom 1.40.0 by manually `apm install` every single dependency listed in [this helpful comment on 22 Aug 2018](https://github.com/platformio/platformio-atom-ide/issues/2059#issuecomment-414883209).
Actually I've `apm install`-ed `platformio-ide-terminal` at the end because I had impression its installation was hanging if done before the rest.

stibinator, 2020-01-21:
On my machine I solved it by installing VSCode. Haven't looked back…

Kawasalto, 2020-04-03:
In fact, you just have to wait. Swings a considerable amount of data, about 1 GB. So wait

BillyGriffiths, 2020-05-19:
> Atom is a deprecated editor. Please use PlatformIo IDE for VSCode => https://platformio.org/platformio-ide

**Atom is still listed here: **  https://platformio.org/install/integration

This is still a problem. My Atom is hanging trying to install dependencies.

Kawasalto, 2020-05-27:
I had to install it again. And Yes, it's hell. The download speed is very low, 200mb was downloaded in 2 days

BillyGriffiths, 2020-05-27:
> I had to install it again. And Yes, it's hell. The download speed is very low, 200mb was downloaded in 2 days

I gave up, uninstalled atom, and installed VScode. Proper Atom support would have been really nice though. Well done for sticking in there and waiting.

bricekevin, 2020-10-27:
i had to install Clang separately but got it to work with the following:

1. Uninstall Atom
2. Instal Xcode Command Line Tools
3. Validate Clang is instaled
4. Instal Atom
5. Instal from Atom platofrom-ide
6. Be Patient....it gets hung for 5-10 mins
7. It will complete 
[image]

kogram, 2020-11-05:
CTRL+T (Stop hangs) next ESC

- Uninstall platformio-ide in Atom
- Install Visual Studio (latest version)

```
apm install platformio-ide-debugger
apm install platformio-ide-terminal
apm install platformio-ide
```