platformio/platformio-atom-ide #2071: Start PIO Home Server

thekansas, 2018-09-14:
# Description of problem
After installation of platformio.ide I get this error two times
**Start PIO Home Server
Traceback (most recent call last): File "main.py", line 17, in init main File "cmd_home.py", line 39, in init cmd_home TypeError: init() got an unexpected keyword argument 'expire'**
I searched the below url but without results.
If I restart atom I get always the same errors.
Waiting your help I send you my best regards
Antonio

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.30.0
PIO IDE: v2.2.2
System: Linux, 4.15.0-34-generic, x64

# Exception
```
Traceback (most recent call last):
  File "main.py", line 17, in init main
  File "cmd_home.py", line 39, in init cmd_home
TypeError: __init__() got an unexpected keyword argument 'expire'

```

thekansas, 2018-09-14:
I went into the terminal window and I try to run "sudo pio upgrade", after that Atom runs w/o error

ivankravets, 2018-09-21:
You have multiple versions of PlatformIO Core in a system. http://docs.platformio.org/en/latest/faq.html#multiple-pio-cores-in-a-system

closed 2018-09-21 by ivankravets