platformio/platformio-atom-ide #2074: Build of Marlin 1.1.9 fails ChopperConfiguration.pde

marclefevre, 2018-09-18:
This is nearly a duplicate of #1802 with the exception that I'm building Marlin 1.1.9 for the melzi platform rather than Marlin_bugfix_2.0.x for an ATMEGA2560 in #1802.  1802 is closed so I assume that I need to open a new issue here.

Here is the error I get:
```
Processing melzi (platform: atmelavr; board: sanguino_atmega1284p; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
LibraryManager: Installing c1921b4
Downloading...
Unpacking...
Error: Traceback (most recent call last):
  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\marc\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__

    return self.main(*args, **kwargs)
  File "c:\users\marc\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\marc\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\marc\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\marc\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\marc\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 107, in cli
    result = (envname, ep.process())
  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 205, in process
    result = self._run()
  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 300, in _run
    self.verbose)
  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 323, in _autoinstall_libdeps
    ctx.invoke(cmd_lib_install, libraries=[lib], silent=not verbose)
  File "c:\users\marc\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\marc\.platformio\penv\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\commands\lib.py", line 103, in lib_install
    library, silent=silent, interactive=interactive, force=force)
  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 354, in install
    force=force)
  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 718, in install
    name, url, requirements, track=True)
  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 444, in _install_from_url
    self.unpack(dlpath, tmp_dir)
  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 222, in unpack
    raise e
IOError: [Errno 2] No such file or directory: 'C:\\Users\\marc\\Documents\\Projects\\3D printing\\AM8\\Marlin AM8_1\\Marlin-1.1.9\\Marlin-1.1.9\\.piolibdeps\\_tmp_installing-qubmkl-package
\\TMC26XStepper-c1921b4856a5ef4b659b661b8cb5dd56f3075e57\\examples\\TMC26XMotorTester\\processing\\TMC26XMotorTest\\ChopperConfiguration.pde'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```
Here is my platformio.ini file:
```
#
# Marlin Firmware
# PlatformIO Configuration File
#
# For detailed documentation with EXAMPLES:
#
# http://docs.platformio.org/en/latest/projectconf.html
#

# Automatic targets - enable auto-uploading
# targets = upload

#
# By default platformio build will abort after 5 errors.
# Remove '-fmax-errors=5' from build_flags below to see all.
#

[platformio]
src_dir     = Marlin
build_dir   = .pioenvs
lib_dir     = .piolib
libdeps_dir = .piolibdeps
env_default = melzi

[common]
build_flags = -fmax-errors=5
  -g
  -ggdb
lib_deps =
  https://github.com/MarlinFirmware/U8glib-HAL/archive/dev.zip
  LiquidCrystal_I2C@1.1.2
  TMC2130Stepper
  https://github.com/teemuatlut/TMC2208Stepper/archive/v0.1.1.zip
  Adafruit NeoPixel@1.1.3
  https://github.com/lincomatic/LiquidTWI2/archive/30aa480.zip
  https://github.com/ameyer/Arduino-L6470/archive/master.zip
  https://github.com/trinamic/TMC26XStepper/archive/c1921b4.zip

#################################
#                               #
#   Unique Core Architectures   #
#                               #
#  Add a new "env" below if no  #
# entry has values suitable to  #
#   build for a given board.    #
#                               #
#################################

#
# ATmega2560
#
[env:megaatmega2560]
platform          = atmelavr
framework         = arduino
board             = megaatmega2560
build_flags       = ${common.build_flags}
board_build.f_cpu = 16000000L
lib_deps          = ${common.lib_deps}
monitor_speed     = 250000

#
# ATmega1280
#
[env:megaatmega1280]
platform          = atmelavr
framework         = arduino
board             = megaatmega1280
build_flags       = ${common.build_flags}
board_build.f_cpu = 16000000L
lib_deps          = ${common.lib_deps}
monitor_speed     = 250000

#
# AT90USB1286 boards using CDC bootloader
# - BRAINWAVE
# - BRAINWAVE_PRO
# - SAV_MKI
# - TEENSYLU
#
[env:at90USB1286_CDC]
platform      = teensy
framework     = arduino
board         = at90USB1286
build_flags   = ${common.build_flags}
lib_deps      = ${common.lib_deps}
lib_ldf_mode  = deep+
extra_scripts = pre:buildroot/share/atom/create_custom_upload_command_CDC.py

#
# AT90USB1286 boards using DFU bootloader
# - PrintrBoard
# - PrintrBoard Rev.F
# - ? 5DPRINT ?
#
[env:at90USB1286_DFU]
platform      = teensy
framework     = arduino
board         = at90USB1286
build_flags   = ${common.build_flags}
lib_deps      = ${common.lib_deps}
lib_ldf_mode  = deep+
extra_scripts = pre:buildroot/share/atom/create_custom_upload_command_DFU.py

#
# Melzi and clones (ATmega1284p)
#
[env:melzi]
platform      = atmelavr
framework     = arduino
board         = sanguino_atmega1284p
build_flags   = ${common.build_flags}
upload_speed  = 57600
lib_deps      = ${common.lib_deps}
monitor_speed = 250000

#
# Melzi and clones (Optiboot bootloader)
#
[env:melzi_optiboot]
platform      = atmelavr
framework     = arduino
board         = sanguino_atmega1284p
build_flags   = ${common.build_flags}
upload_speed  = 115200
lib_deps      = ${common.lib_deps}
monitor_speed = 250000

#
# RAMBo
#
[env:rambo]
platform          = atmelavr
framework         = arduino
board             = reprap_rambo
build_flags       = ${common.build_flags}
board_build.f_cpu = 16000000L
lib_deps          = ${common.lib_deps}
monitor_speed     = 250000

#
# Sanguinololu (ATmega644p)
#
[env:sanguino_atmega644p]
platform      = atmelavr
framework     = arduino
board         = sanguino_atmega644p
build_flags   = ${common.build_flags}
lib_deps      = ${common.lib_deps}
monitor_speed = 250000

#
# Sanguinololu (ATmega1284p)
#
[env:sanguino_atmega1284p]
platform      = atmelavr
framework     = arduino
board         = sanguino_atmega1284p
build_flags   = ${common.build_flags}
lib_deps      = ${common.lib_deps}
monitor_speed = 250000

#
# Teensy++ 2.0
#
[env:teensy20]
platform          = teensy
framework         = arduino
board             = teensy20pp
build_flags       = ${common.build_flags}
#board_build.f_cpu = 20000000L     ; Bug in Arduino framework disallows boards running at 20Mhz
lib_deps          = ${common.lib_deps}
lib_ldf_mode      = deep+
monitor_speed     = 250000
```
This is on a 64-bit windows 10 installation running under Atom.  The Platformio version is 

Home: 1.0.2
Core: 3.6.1a4

ivankravets, 2018-09-21:
1. Please try to disable antivirus software which could block our library installer
2. Try [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide).

See my comment https://github.com/MarlinFirmware/MarlinDocumentation/issues/176

closed 2018-09-21 by ivankravets

pedroamfarias, 2019-04-20:
This topic is closed but the same issue happened with me today. I solve it just copy the project to root (like c:\platformio\project\...) instead of keeping inside long Windows path. Looks like putting the project inside long path generate this issue =)

aalku, 2019-10-08:
Right. Has something to do with paths too long or something. Thank you, @pedroamfarias.