platformio/platformio-atom-ide #2084: pioplus fails to run linux_armv7l

cdzieciol, 2018-10-04:
<h2> Configuration </h2>
ubuntu@ubuntu:~$ uname -a
Linux ubuntu 4.15.0-1024-raspi2 #26-Ubuntu SMP PREEMPT Tue Sep 25 06:39:44 UTC 2018 armv7l armv7l armv7l GNU/Linux

ubuntu@ubuntu:~$ pio update
Updating tool-scons                      @ 2.20501.4      [Up-to-date]
Updating contrib-pysite                  @ 0.3.2          [Up-to-date]
Updating tool-pioplus                    @ 1.4.10         [Up-to-date]

Platform Manager
================
Platform Atmel AVR
--------
Updating atmelavr                        @ 1.10.1         [Up-to-date]
Updating toolchain-atmelavr              @ 1.40902.0      [Up-to-date]
Updating framework-arduinoavr            @ 1.10621.1      [Up-to-date]
Updating tool-avrdude                    @ 1.60300.2      [Up-to-date]

Library Manager
===============
Updating CONTROLLINO                     @ 3.0.2          [Up-to-date]

ubuntu@ubuntu:~$ pio upgrade
You're up-to-date!
PlatformIO 3.6.0 is currently the newest version available.

<h2>Exception</h2>

ubuntu@ubuntu:~$ pio home
Error: Traceback (most recent call last):
  File "/home/ubuntu/.local/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/home/ubuntu/.local/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/home/ubuntu/.local/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/home/ubuntu/.local/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/ubuntu/.local/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/ubuntu/.local/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/home/ubuntu/.local/lib/python2.7/site-packages/platformio/commands/home.py", line 32, in cli
    pioplus_call(sys.argv[1:])
  File "/home/ubuntu/.local/lib/python2.7/site-packages/platformio/managers/core.py", line 114, in pioplus_call
    code = subprocess.call([pioplus_path] + args, **kwargs)
  File "/usr/lib/python2.7/subprocess.py", line 172, in call
    return Popen(*popenargs, **kwargs).wait()
  File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory

ubuntu@ubuntu:~/.platformio/packages/tool-pioplus$ ./pioplus
-bash: ./pioplus: No such file or directory

cdzieciol, 2018-10-10:
ubuntu@ubuntu:~$ file ~/.platformio/packages/tool-pioplus/pioplus
/home/ubuntu/.platformio/packages/tool-pioplus/pioplus: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=966936a0bb7822079126640c59553cb36fc23444, stripped

ubuntu@ubuntu:~$ ldd ~/.platformio/packages/tool-pioplus/pioplus
	not a dynamic executable

ivankravets, 2018-10-29:
Fixed in the latest release. Please run in terminal `pio upgrade`.

closed 2018-10-29 by ivankravets