platformio/platformio-atom-ide #2088: InstallationManager

HamzaHajeir, 2018-10-06:
Hi 

I got this error while installing platformio, 
my python is 2.7.15, installed LLVM followed instructions.

> Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv at t. (C:\Users\Hamza.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1:39576) at x (C:\Users\Hamza.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1:20834) at Generator._invoke (C:\Users\Hamza.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1:20622) at Generator.e.(anonymous function) [as throw] (C:\Users\Hamza.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1:21013) at n (C:\Users\Hamza.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1:32216) at C:\Users\Hamza.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1:32335 at at process._tickCallback (internal/process/next_tick.js:188:7)
> Report a problem
> 

Tried my best hard working to manually creating it, but i failed..

see my progress : 

> PS C:\Users\Hamza\.atom> virtualenv --help
> virtualenv : The term 'virtualenv' is not recognized as the name of a cmdlet, function,
> script file, or operable program. Check the spelling of the name, or if a path was included,
> verify that the path is correct and try again.
> At line:1 char:1
> + virtualenv --help
> + ~~~~~~~~~~
>     + CategoryInfo          : ObjectNotFound: (virtualenv:String) [], CommandNotFoundExceptio
>    n
>     + FullyQualifiedErrorId : CommandNotFoundException
> 
> 
> PS C:\Users\Hamza\.atom> pip install virtualenv
> Program 'pip.exe' failed to run: Access is deniedAt line:1 char:1
> + pip install virtualenv
> + ~~~~~~~~~~~~~~~~~~~~~~.
> At line:1 char:1
> + pip install virtualenv
> + ~~~~~~~~~~~~~~~~~~~~~~
>     + CategoryInfo          : ResourceUnavailable: (:) [], ApplicationFailedException
>     + FullyQualifiedErrorId : NativeCommandFailed
> 
> PS C:\Users\Hamza\.atom> pip2 install virtualenv
> Program 'pip2.exe' failed to run: Access is deniedAt line:1 char:1
> + pip2 install virtualenv
> + ~~~~~~~~~~~~~~~~~~~~~~~.
> At line:1 char:1
> + pip2 install virtualenv
> + ~~~~~~~~~~~~~~~~~~~~~~~
>     + CategoryInfo          : ResourceUnavailable: (:) [], ApplicationFailedException
>     + FullyQualifiedErrorId : NativeCommandFailed
> 
> PS C:\Users\Hamza\.atom> curl https://bootstrap.pypa.io/get-pip.py -o get-pip.py
> >>
> curl : The request was aborted: Could not create SSL/TLS secure channel.
> At line:1 char:1
> + curl https://bootstrap.pypa.io/get-pip.py -o get-pip.py
> + ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
>     + CategoryInfo          : InvalidOperation: (System.Net.HttpWebRequest:HttpWebRequest) [I
>    nvoke-WebRequest], WebException
>     + FullyQualifiedErrorId : WebCmdletWebResponseException,Microsoft.PowerShell.Commands.Inv
>    okeWebRequestCommand

HamzaHajeir, 2018-10-09:
I've solved it : 

somethings i didn't know  : 
1- where python codes are extracting . ( I thought at python shell and so on ) -> Windows cmd .
2- how to run python code : 
because writing python commands will not be understood by cmd, So before every command, write : python -m .

3- at [official manual installation guide](http://docs.platformio.org/en/latest/installation.html#virtual-environment) and at step 2 in creating - Activate virtual environment. 
It has not a correct cmd. 
The activation file should be .py extended, Which found in penv\Scripts folder , In my case it's named "activate_this.py" .

So command line should be as like : C:\Users\Username\.platformio\penv\Scripts\activate_this.py

ivankravets, 2018-10-10:
Do you have `C:\Users\Username.platformio\penv\Scripts\activate.bat`?

HamzaHajeir, 2018-10-10:
@ivankravets  Yeah i have it, I later activated it.

One notation : after installing some plugins and so, /penv folder changed its name to /New Folder . So atom tried to install PIO Core and showed up the error again. 

I followed the installation guide again and it worked and created a new penv folder. 
after that i started atom again, and new penv folder has been deleted xD . After that i noted /New folder is the old /penv, renamed it back to penv and till now it's working without problems.

ivankravets, 2018-10-12:
> now it's working without problems.

Happy coding with PlatformIO! ;)

closed 2018-10-12 by ivankravets