platformio/platformio-atom-ide #2105: InstallationManager

Lyukvoice, 2018-11-18:
Error: Failed to download file https://github.com/platformio/platformio/archive/develop.zip: 
    at _callee$ (C:\Users\lyukv\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:2679:19)
    at tryCatch (C:\Users\lyukv\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:836:40)
    at Generator.invoke [as _invoke] (C:\Users\lyukv\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1070:22)
    at Generator.prototype.(anonymous function) [as next] (C:\Users\lyukv\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:888:21)
    at step (C:\Users\lyukv\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:2781:191)
    at C:\Users\lyukv\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:2781:361
    at <anonymous>

ivankravets, 2018-11-29:
Please update to 2.3.0 version. Does it work now?

SebDominguez, 2019-03-11:
Just trying to add the platformio-ide package.
`apm install platformio-ide` is running endlessly, also same thing using the GUI.

Atom 1.34.0
Platformio-ide 2.3.1
macOS 10.14.3

ivankravets, 2019-07-23:
Please switch to PlatformIO IDE for VSCode =>  https://platformio.org/platformio-ide

Does it work here?

closed 2019-07-23 by ivankravets