platformio/platformio-atom-ide #2106: Start PIO Home Server

pradipshenolkar, 2018-11-23:
# Description of problem
Why does this error occur?

# Configuration
Atom: 1.32.1
PIO IDE: v2.2.5
System: Windows_NT, 6.3.9600, x64

# Exception
```
Error: 'platformio' is not recognized as an internal or external command,
operable program or batch file.
Error: spawn platformio ENOENT
    at C:\Users\Pradip's\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:15518
    at h (C:\Users\Pradip's\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:5944)
    at ChildProcess.r.on.e (C:\Users\Pradip's\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:5733)
    at emitOne (events.js:116:13)
    at ChildProcess.emit (events.js:211:7)
    at ChildProcess.cp.emit (C:\Users\Pradip's\.atom\packages\platformio-ide\node_modules\cross-spawn\lib\enoent.js:30:37)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
```

ivankravets, 2018-11-29:
Could you update to the latest 2.3.0 version? Does it work now?

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets