platformio/platformio-atom-ide #2107: Failed to activate the platformio-ide package

workino, 2018-11-26:
**Atom**: 1.18.0 x64
**Electron**: 1.3.15
**OS**: "Red Hat
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.2.6

### Stack Trace

Failed to activate the platformio-ide package

```
At Unexpected token function

SyntaxError: Unexpected token function
    at Module.get_Module._compile (/usr/share/atom/resources/app/src/native-compile-cache.js:94:52)
    at Object.value [as .js] (/usr/share/atom/resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (usr/share/atom/resources/app/static/index.js:47:45)
    at require (/usr/share/atom/resources/app/src/native-compile-cache.js:66:33)
    at /packages/platformio-ide/lib/maintenance.js:24:30)
    at Module.get_Module._compile (/usr/share/atom/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/usr/share/atom/resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (usr/share/atom/resources/app/static/index.js:47:45)
    at require (/usr/share/atom/resources/app/src/native-compile-cache.js:66:33)
    at /packages/platformio-ide/lib/main.js:11:19)
    at Module.get_Module._compile (/usr/share/atom/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/usr/share/atom/resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (usr/share/atom/resources/app/static/index.js:47:45)
    at require (internal/module.js:20:19)
    at customRequire (/usr/share/atom/resources/app/static/<embedded>:96:26)
    at Package.module.exports.Package.requireMainModule (/usr/share/atom/resources/app/src/package.js:887:35)
    at Package.module.exports.Package.activateNow (/usr/share/atom/resources/app/src/package.js:242:22)
    at /usr/share/atom/resources/app/src/package.js:1014:34
    at Function.module.exports.Emitter.simpleDispatch (/usr/share/atom/resources/app/node_modules/event-kit/lib/emitter.js:25:20)
    at Emitter.module.exports.Emitter.emit (/usr/share/atom/resources/app/node_modules/event-kit/lib/emitter.js:141:34)
    at /usr/share/atom/resources/app/src/package-manager.js:658:56
    at Set.forEach (native)
    at PackageManager.module.exports.PackageManager.activatePackage (/usr/share/atom/resources/app/src/package-manager.js:656:47)
    at /usr/share/atom/resources/app/node_modules/settings-view/lib/package-manager.js:519:35
    at exit (/usr/share/atom/resources/app/node_modules/settings-view/lib/package-manager.js:128:22)
    at triggerExitCallback (/usr/share/atom/resources/app/src/buffered-process.js:338:17)
    at ChildProcess.<anonymous> (/usr/share/atom/resources/app/src/buffered-process.js:368:17)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
```

### Commands

```
```

### Non-Core Packages

```
platformio-ide 2.2.6 
```

demonskull, 2018-11-29:
Yep, same here

ivankravets, 2018-11-29:
Could you update to the latest 2.3.0 version? Does it work now?

miknuc, 2018-12-09:
Yes, I tried with 2.3.0 and got the same issue (ubuntu 18.10, atom 1.15.0 x64)

woergi, 2018-12-11:
Same for me for:
```
Linux Mint 17.3 64Bit
Atom 1.14.4
```

After updating Atom to `1.26.1` the issue disappeared.

neuman1812, 2019-07-19:
I know this is kinda old,  but its happening to me

Ubuntu 18.04  64 Bit

Atom 1.26.1

The menu shows,  but no functions work in the drop down.

~~~~~~~~~~~~~~~~~~~~~~~~ 

[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.26.1 x64
**Electron**: 1.7.11
**OS**: Ubuntu 18.04.2
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.4.1

### Stack Trace

Failed to activate the platformio-ide package

```
At Unexpected token ...

SyntaxError: Unexpected token ...
    at Module.get_Module._compile (/opt/atom/resources/app/src/native-compile-cache.js:94:52)
    at Object.value [as .js] (/opt/atom/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (/opt/atom/resources/app/src/native-compile-cache.js:66:33)
    at /packages/platformio-ide/node_modules/ws/index.js:3:19)
    at /packages/platformio-ide/node_modules/ws/index.js:12:3)
    at Module.get_Module._compile (/opt/atom/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/opt/atom/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (/opt/atom/resources/app/src/native-compile-cache.js:66:33)
    at /packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:422)
    at /packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:1080)
    at /packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:3:3)
    at Module.get_Module._compile (/opt/atom/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/opt/atom/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (/opt/atom/resources/app/src/native-compile-cache.js:66:33)
    at /packages/platformio-ide/lib/maintenance.js:24:30)
    at /packages/platformio-ide/lib/maintenance.js:131:2)
    at Module.get_Module._compile (/opt/atom/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/opt/atom/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (/opt/atom/resources/app/src/native-compile-cache.js:66:33)
    at /packages/platformio-ide/lib/main.js:11:19)
    at /packages/platformio-ide/lib/main.js:334:22)
    at Module.get_Module._compile (/opt/atom/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/opt/atom/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (internal/module.js:20:19)
    at customRequire (/opt/atom/resources/app/static/<embedded>:96:26)
    at Package.requireMainModule (/opt/atom/resources/app/src/package.js:782:33)
    at Package.activateNow (/opt/atom/resources/app/src/package.js:220:40)
    at activationHookSubscriptions.add.packageManager.onDidTriggerActivationHook (/opt/atom/resources/app/src/package.js:901:81)
    at Function.module.exports.Emitter.simpleDispatch (/opt/atom/resources/app/node_modules/event-kit/lib/emitter.js:25:20)
    at Emitter.module.exports.Emitter.emit (/opt/atom/resources/app/node_modules/event-kit/lib/emitter.js:141:34)
    at triggeredActivationHooks.forEach.hook (/opt/atom/resources/app/src/package-manager.js:705:86)
    at Set.forEach (native)
    at PackageManager.activatePackage (/opt/atom/resources/app/src/package-manager.js:705:43)
    at /opt/atom/resources/app/node_modules/settings-view/lib/package-manager.js:491:35
    at exit (/opt/atom/resources/app/node_modules/settings-view/lib/package-manager.js:128:22)
    at triggerExitCallback (/opt/atom/resources/app/src/buffered-process.js:267:15)
    at bufferStream (/opt/atom/resources/app/src/buffered-process.js:288:15)
    at Socket.stream.on (/opt/atom/resources/app/src/buffered-process.js:164:13)
    at emitOne (events.js:101:20)
    at Socket.emit (events.js:191:7)
    at Pipe._handle.close [as _onclose] (net.js:510:12)
```

### Commands

```
     -3:18.3.0 platformio-ide:help.report-platformio-issue (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -3:12.5.0 platformio-ide:maintenance.serial-monitor (input.hidden-input)
     -3:10.5.0 platformio-ide:maintenance.serial-ports (input.hidden-input)
     -2:51.8.0 platformio-ide:home (input.hidden-input)
     -2:43.2.0 platformio-ide:help-website (input.hidden-input)
     -0:54.9.0 core:paste (input.hidden-input)
     -0:53.6.0 editor:move-to-first-character-of-line (input.hidden-input)
 23x -0:53.4.0 core:delete (input.hidden-input)
     -0:51.4.0 editor:move-to-end-of-screen-line (input.hidden-input)
 12x -0:51.1.0 core:backspace (input.hidden-input)
     -0:48.8.0 core:confirm (input.hidden-input)
```

### Non-Core Packages

```
platformio-ide 2.4.1 
platformio-ide-debugger 1.2.6 
```

jaekar99, 2019-07-19:
New for me.. Linux Mint 19.1

[Enter steps to reproduce:]

Reboot of IDE and... Error occurs.
2. ...

**Atom**: 1.26.1 x64
**Electron**: 1.7.11
**OS**: Linux Mint
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.4.1

### Stack Trace

Failed to activate the platformio-ide package

```
At Unexpected token ...

SyntaxError: Unexpected token ...
    at Module.get_Module._compile (/opt/atom/resources/app/src/native-compile-cache.js:94:52)
    at Object.value [as .js] (/opt/atom/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (/opt/atom/resources/app/src/native-compile-cache.js:66:33)
    at /packages/platformio-ide/node_modules/ws/index.js:3:19)
    at /packages/platformio-ide/node_modules/ws/index.js:12:3)
    at Module.get_Module._compile (/opt/atom/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/opt/atom/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (/opt/atom/resources/app/src/native-compile-cache.js:66:33)
    at /packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:422)
    at /packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:1080)
    at /packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:3:3)
    at Module.get_Module._compile (/opt/atom/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/opt/atom/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (/opt/atom/resources/app/src/native-compile-cache.js:66:33)
    at /packages/platformio-ide/lib/maintenance.js:24:30)
    at /packages/platformio-ide/lib/maintenance.js:131:2)
    at Module.get_Module._compile (/opt/atom/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/opt/atom/resources/app/src/compile-cache.js:240:29)
```

### Commands

```
     -3:09.5.0 core:confirm (input.hidden-input)
```

### Non-Core Packages

```
autocomplete-clang 0.12.0 
build 0.70.0 
busy-signal 2.0.1 
file-icons 2.1.33 
intentions 1.1.5 
language-ini 1.21.0 
linter 2.3.0 
linter-gcc 0.7.1 
linter-ui-default 1.7.1 
platformio-ide 2.4.1 
platformio-ide-debugger 1.2.6 
platformio-ide-terminal 2.9.1 
tool-bar 1.1.12 
```

ivankravets, 2019-07-19:
Could you update Atom to the latest version?

In any case, we highly recommend switching to PlatformIO IDE for VSCode https://platformio.org/platformio-ide

jaekar99, 2019-07-19:
Full uninstall and reinstall from command line from instructions here:
https://computingforgeeks.com/installing-atom-text-editor-on-ubuntu-18-04-linux-mint-1
Working great now.

neuman1812, 2019-07-19:
Updating worked.  I was using the ppa for atom from webupd8team.  They are a few updates behind.   Changed over to Packagecloud and got the update to 1.38.2.   PlatformIO installed with that.

Thanks

DanielFLopez, 2019-07-31:
I currently have this error with the latest version of the Atom, I tried with **apm install platformio/platformio-atom-ide** by console and reinstalling Atom but doesn't works. 

```Cannot find module './pubsuffix-psl'
Error: Cannot find module './pubsuffix-psl'
    at Module._resolveFilename (internal/modules/cjs/loader.js:602:15)
    at Module._resolveFilename (C:\Users\dlopez\AppData\Local\atom\app-1.39.1\resources\electron.asar\common\reset-search-paths.js:35:12)
    at Function.get_Module._resolveFilename (C:\Users\dlopez\AppData\Local\atom\app-1.39.1\resources\app\static\<embedded>:11:155394)
    at Module.require (file:///C:/Users/dlopez/AppData/Local/atom/app-1.39.1/resources/app.asar/static/index.js:60:43)
    at require (C:\Users\dlopez\AppData\Local\atom\app-1.39.1\resources\app\static\<embedded>:11:146719)
    at Object.<anonymous> (C:\Users\dlopez\.atom\packages\platformio-ide\node_modules\tough-cookie\lib\cookie.js:35:17)
    at Object.<anonymous> (C:\Users\dlopez\.atom\packages\platformio-ide\node_modules\tough-cookie\lib\cookie.js:1433:3)
    at Module.get_Module._compile (C:\Users\dlopez\AppData\Local\atom\app-1.39.1\resources\app\static\<embedded>:11:147410)
    at Object.value [as .js] (C:\Users\dlopez\AppData\Local\atom\app-1.39.1\resources\app\static\<embedded>:11:150958)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (file:///C:/Users/dlopez/AppData/Local/atom/app-1.39.1/resources/app.asar/static/index.js:71:46)
    at require (C:\Users\dlopez\AppData\Local\atom\app-1.39.1\resources\app\static\<embedded>:11:146719)
    at Object.<anonymous> (C:\Users\dlopez\.atom\packages\platformio-ide\node_modules\request\lib\cookies.js:3:13)
    at Object.<anonymous> (C:\Users\dlopez\.atom\packages\platformio-ide\node_modules\request\lib\cookies.js:40:3)
    at Module.get_Module._compile (C:\Users\dlopez\AppData\Local\atom\app-1.39.1\resources\app\static\<embedded>:11:147410)
    at Object.value [as .js] (C:\Users\dlopez\AppData\Local\atom\app-1.39.1\resources\app\static\<embedded>:11:150958)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (file:///C:/Users/dlopez/AppData/Local/atom/app-1.39.1/resources/app.asar/static/index.js:71:46)
    at require (C:\Users\dlopez\AppData\Local\atom\app-1.39.1\resources\app\static\<embedded>:11:146719)
    at Object.<anonymous> (C:\Users\dlopez\.atom\packages\platformio-ide\node_modules\request\index.js:18:15)
    at Object.<anonymous> (C:\Users\dlopez\.atom\packages\platformio-ide\node_modules\request\index.js:157:3)
    at Module.get_Module._compile (C:\Users\dlopez\AppData\Local\atom\app-1.39.1\resources\app\static\<embedded>:11:147410)
    at Object.value [as .js] (C:\Users\dlopez\AppData\Local\atom\app-1.39.1\resources\app\static\<embedded>:11:150958)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
```

gufonero, 2019-08-14:
This error is fixed in recent versions; for me it disappeared when I updated to v.1.40.0 (Linux Mint 19.1 environment). 
To get the latest versions you need to add the packagecloud repository to your apt sources as suggested by @jaekar99 above (see the link pointed to in that message for instructions).
One possible problem is that you may have installed Atom manually in a different directory (other than `/usr/bin` which is where apt puts it) and you are starting the wrong version. Type `which atom` in a terminal session to check where you are starting it from. If the answer is not `/usr/bin/atom` this may be your problem.

okpail, 2019-12-22:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.18.0 x64
**Electron**: 1.3.15
**OS**: Mac OS X 10.15.2
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.6.0

### Stack Trace

Failed to activate the platformio-ide package

```
At Unexpected token function

SyntaxError: Unexpected token function
    at Module.get_Module._compile (/Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/src/native-compile-cache.js:94:52)
    at Object.value [as .js] (/Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (/Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at /packages/platformio-ide/lib/maintenance.js:24:30)
    at Module.get_Module._compile (/Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (/Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at /packages/platformio-ide/lib/main.js:11:19)
    at Module.get_Module._compile (/Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (internal/module.js:20:19)
    at customRequire (/Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/static/<embedded>:96:26)
    at Package.module.exports.Package.requireMainModule (/Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/src/package.js:887:35)
    at Package.module.exports.Package.activateNow (/Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/src/package.js:242:22)
    at /Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/src/package.js:1014:34
    at Function.module.exports.Emitter.simpleDispatch (/Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/node_modules/event-kit/lib/emitter.js:25:20)
    at Emitter.module.exports.Emitter.emit (/Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/node_modules/event-kit/lib/emitter.js:141:34)
    at /Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/src/package-manager.js:658:56
    at Set.forEach (native)
    at PackageManager.module.exports.PackageManager.activatePackage (/Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/src/package-manager.js:656:47)
    at /packages/settings-view/lib/package-manager.coffee:351:25
    at exit (/packages/settings-view/lib/package-manager.coffee:68:7)
    at triggerExitCallback (/Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/src/buffered-process.js:338:17)
    at /Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/src/buffered-process.js:359:17
    at Socket.<anonymous> (/Applications/LunaStudio.app/Contents/Resources/1.9/third-party/Atom.app/Contents/Resources/app/src/buffered-process.js:220:15)
    at emitOne (events.js:101:20)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:493:12)
```

### Commands

```
  3x -4:20 dev-live-reload:reload-all (atom-workspace.workspace.scrollbars-visible-when-scrolling)
  2x -3:45.3.0 tree-view:reveal-active-file (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-luna-syntax.theme-luna-dark-ui)
 15x -2:34.5.0 core:backspace (input.hidden-input)
     -2:29.6.0 core:confirm (input.hidden-input)
```

### Non-Core Packages

```
column-select 0.2.0 
cursor-history 0.11.1 
hasklig 0.4.0 
highlight-line 0.12.0 
highlight-selected 0.13.1 
intentions 1.1.5 
luna-dark-ui 0.1.1 
luna-dpi 0.1.2 
luna-studio 1.0.0 
luna-syntax 0.1.0 
minimap 4.29.6 
minimap-cursorline 0.2.0 
minimap-find-and-replace 4.5.2 
minimap-highlight-selected 4.6.1 
multi-cursor 2.1.5 
platformio-ide 2.6.0 
sort-lines 0.14.0 
```

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets