platformio/platformio-atom-ide #2108: InstallationManager

yguel, 2018-11-26:
Error: Failed to download file https://github.com/platformio/platformio/archive/develop.zip: 
    at _callee$ (/home/manu/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2668:19)
    at tryCatch (/home/manu/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:836:40)
    at Generator.invoke [as _invoke] (/home/manu/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1070:22)
    at Generator.prototype.(anonymous function) [as next] (/home/manu/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:888:21)
    at step (/home/manu/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2770:191)
    at /home/manu/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2770:361

ivankravets, 2018-11-29:
Could you update to the latest 2.3.0 version? Does it work now?

ivankravets, 2018-11-29:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/2105

closed 2018-11-29 by ivankravets