platformio/platformio-atom-ide #2111: InstallationManager

zenbox, 2018-11-30:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.32.2
PIO IDE: v2.3.0
System: Darwin, 18.2.0, x64

# Exception
```
Error: PIP Core: /Users/michaelreichart/.platformio/penv/bin/python2.7: No module named pip

    at i.runCommand (/Users/michaelreichart/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:26217)
    at ChildProcess.h (/Users/michaelreichart/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:5916)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at maybeClose (internal/child_process.js:925:16)
    at Socket.stream.socket.on (internal/child_process.js:346:11)
    at emitOne (events.js:116:13)
    at Socket.emit (events.js:211:7)
    at Pipe._handle.close [as _onclose] (net.js:554:12)
```

DeviousPenguin, 2018-11-30:
I'm getting the same thing, here's my log:

```
Error: PIP Core: /Users/mmmw/.platformio/penv/bin/python2.7: No module named pip

    at i.runCommand (/Users/mmmw/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:26217)
    at ChildProcess.h (/Users/mmmw/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:5916)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at maybeClose (internal/child_process.js:925:16)
    at Socket.stream.socket.on (internal/child_process.js:346:11)
    at emitOne (events.js:116:13)
    at Socket.emit (events.js:211:7)
    at Pipe._handle.close [as _onclose] (net.js:554:12)
```

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets