platformio/platformio-atom-ide #2127: Error buidling of a new added library to platformio.

HamzaHajeir, 2018-12-20:
Hello guys.

I've [asked](https://github.com/platformio/platformio-libmirror/issues/146) yesterday adding ESPArto library and its dependent libraries. Thankfully responded with addition . 

Now I'm trying to build the first example at it, Faced some errors . 

I've included all libraries to project, defined them in platformio.ini under lib-deps: 

building example : "01_Simple"
but i have the following error at building process : 

[image]

*The first error (missing library) is noted before buidling.

It's working well at Arduino IDE. 
I'm using Atom, PlatformIO is updated.

Note : I can't copy verbose terminal.

HamzaHajeir, 2018-12-29:
How can i even copy all the build panel verbose ?

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets

HamzaHajeir, 2020-06-14:
> We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.
>
> **Please note that we HIGHLY recommend moving to [PlatformIO IDE [...]

Thanks, I had solved it since then, sorry for not closing it.