platformio/platformio-atom-ide #2172: Uninstall incompatible packages: platformio-ide-terminal

ganiutaiwo, 2019-04-01:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.34.0
PIO IDE: v2.3.1
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Failed to delete platformio-ide-terminal: EPERM: operation not permitted, lstat 'C:\Users\Taiwo\.atom\packages\platformio-ide-terminal\node_modules\pty.js\bin\win32\ia32_m49'

    at file:///C:/Users/Taiwo/.atom/packages/platformio-ide/lib/installer/stages/atom-dependencies.js:139:25
    at file:///C:/Users/Taiwo/.atom/packages/platformio-ide/lib/utils.js:67:7
    at ChildProcess.h (C:\Users\Taiwo\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:5916)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at ChildProcess.cp.emit (C:\Users\Taiwo\.atom\packages\platformio-ide\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:925:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
```

ganiutaiwo, 2019-04-01:
my terminal is not working

ivankravets, 2019-07-23:
Please switch to PlatformIO IDE for VSCode => https://platformio.org/platformio-ide

Does it work here?

closed 2019-07-23 by ivankravets