platformio/platformio-atom-ide #2184: Start PIO Home Server

mazdah, 2019-04-19:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.36.0
PIO IDE: v2.3.1
System: Darwin, 18.5.0, x64

# Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

    at /Users/mazdah/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:15270
    at ChildProcess.h (/Users/mazdah/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:5904)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at maybeClose (internal/child_process.js:925:16)
    at Socket.stream.socket.on (internal/child_process.js:346:11)
    at emitOne (events.js:116:13)
    at Socket.emit (events.js:211:7)
    at Pipe._handle.close [as _onclose] (net.js:554:12)
```

exocode, 2019-04-19:
Have the same issue:

@ivankravets I **completely deactivated** my Firewall: 
I dont know what to do else... 

I uninstalled EVERYTHING:
- platformio
- conda
- python3.x
- pip
- pip3
....
... 
except `/usr/bin/python` which is part of MacOS (Python 2.7.10) 

**Reinstalled:**
`sudo python -m ensurepip`
`sudo pip install --upgrade pip`
No Anaconda installed till now
`python --version`

I opened VSCode and installed the PlatformIO Plugin, wait, and restarted the IDE. But have 
"Error: Detected unknown package `contrib-pysite`"

```
> platformio --version                                                                                                                                                               
PlatformIO, version 3.6.6

> which platformio                                                                                                                                                                     
/Users/janjezek/.platformio/penv/bin/platformio

> platformio upgrade                                                                                                                                                                    
You're up-to-date!
PlatformIO 3.6.6 is currently the newest version available.
 
> platformio account show                                                                                                                                                               
CorePackageManager: Installing contrib-pysite @ ~2.27.0
Error: Detected unknown package 'contrib-pysite'

```

Error: Detected unknown package `contrib-pysite`

What else could I do?

ivankravets, 2019-04-19:
Duplicate of https://github.com/platformio/platformio-core/issues/2341

closed 2019-04-19 by ivankravets