platformio/platformio-atom-ide #2201: Start PIO Home Server

irbigturtle, 2019-05-28:
# Description of problem
Trying to start the pio home server - fails after a few seconds. 

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.37.0
PIO IDE: v2.3.1
System: Windows_NT, 10.0.17763, x64

# Exception
```
Error: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<string>", line 38, in <module>
  File "<string>", line 39, in <module>
  File "<string>", line 18, in load_module
  File "<string>", line 23, in <module>
ImportError: cannot import name get_project_test_dir

    at C:\Users\Christian\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:14992
    at ChildProcess.h (C:\Users\Christian\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:5916)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at ChildProcess.cp.emit (C:\Users\Christian\.atom\packages\platformio-ide\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:925:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
```

LemarinelNet, 2019-05-28:
Same for me, old installation, was working a few days ago !

ivankravets, 2019-05-28:
What is your PlatformIO Core version? 
Please open PIO IDE Terminal and type: 
`pio —version`

LemarinelNet, 2019-05-28:
PlatformIO, version 4.0.0a14

ivankravets, 2019-05-28:
Please open PlatformIO IDE Terminal and type
```
pio upgrade --dev
```

Restart Atom.

drzacek, 2019-05-29:
Would be great if I could open the terminal, but I can't. Seems to not work too. Don't know how to get any verbose output on that though.

ivankravets, 2019-05-29:
Atom > Menu > PlatformIO > Upgrade PlatformIO Core
[image]

drzacek, 2019-05-29:
[image]

Yes, platformio-ide-terminal is installed. Tried installing, deinstalling, restarting. (Edit->Prefferences->Install)

LemarinelNet, 2019-05-29:
> Please open PlatformIO IDE Terminal and type
> 
> ```
> pio upgrade --dev
> ```
> 
> Restart IDE.

Worked for me. Thanks

drzacek, 2019-05-29:
Don't ask me what I did, I reinstalled many many times and it kicked in. 
Did pio upgrade --dev though and it seem to work now.

closed 2019-07-23 by ivankravets