platformio/platformio-atom-ide #2202: Start PIO Home Server

sergey-k34, 2019-05-28:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.31.2
PIO IDE: v2.3.1
System: Linux, 4.15.0-50-generic, x64

# Exception
```
Error: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<string>", line 38, in <module>
  File "<string>", line 39, in <module>
  File "<string>", line 18, in load_module
  File "<string>", line 23, in <module>
ImportError: cannot import name get_project_test_dir

    at /home/sergei/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:14992
    at ChildProcess.h (/home/sergei/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:5916)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at maybeClose (internal/child_process.js:925:16)
    at Socket.stream.socket.on (internal/child_process.js:346:11)
    at emitOne (events.js:116:13)
    at Socket.emit (events.js:211:7)
    at Pipe._handle.close [as _onclose] (net.js:554:12)
```

ivankravets, 2019-05-28:
What is your version of PIO Core? Is this a new installation?

ivankravets, 2019-05-28:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/2201

closed 2019-05-28 by ivankravets