platformio/platformio-atom-ide #2203: Start PIO Home Server

cloeffler, 2019-05-28:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.37.0
PIO IDE: v2.3.1
System: Windows_NT, 10.0.17763, x64

# Exception
```
Error: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<string>", line 38, in <module>
  File "<string>", line 39, in <module>
  File "<string>", line 18, in load_module
  File "<string>", line 23, in <module>
ImportError: cannot import name get_project_test_dir

    at C:\Users\cloeffler.BSILV\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:14992
    at ChildProcess.h (C:\Users\cloeffler.BSILV\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:5916)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at ChildProcess.cp.emit (C:\Users\cloeffler.BSILV\.atom\packages\platformio-ide\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:925:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
```

symonty, 2019-05-29:
platform io upgrade fixed it for me.

ivankravets, 2019-05-29:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/2201

closed 2019-05-29 by ivankravets