platformio/platformio-atom-ide #2210: InstallationManager

augiem94, 2019-06-01:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.37.0
PIO IDE: v2.3.1
System: Windows_NT, 10.0.17763, x64

# Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: If you use Antivirus, it can block PlatformIO Installer. Try to disable it for a while.

Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\augie\.platformio\.cache\tmp-19248DaK8MmPpmHpX\virtualenv-16.1.0\src\virtualenv.py", line 2462, in <module>
    main()
  File "C:\Users\augie\.platformio\.cache\tmp-19248DaK8MmPpmHpX\virtualenv-16.1.0\src\virtualenv.py", line 762, in main
    symlink=options.symlink,
  File "C:\Users\augie\.platformio\.cache\tmp-19248DaK8MmPpmHpX\virtualenv-16.1.0\src\virtualenv.py", line 1015, in create_environment
    install_wheel(to_install, py_executable, search_dirs, download=download)
  File "C:\Users\augie\.platformio\.cache\tmp-19248DaK8MmPpmHpX\virtualenv-16.1.0\src\virtualenv.py", line 968, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "C:\Users\augie\.platformio\.cache\tmp-19248DaK8MmPpmHpX\virtualenv-16.1.0\src\virtualenv.py", line 812, in call_subprocess
    env=env,
  File "C:\Users\augie\.platformio\python27\Lib\subprocess.py", line 390, in __init__
    errread, errwrite)
  File "C:\Users\augie\.platformio\python27\Lib\subprocess.py", line 640, in _execute_child
    startupinfo)
WindowsError: [Error 5] Access is denied

Please use the *system* python to run this script
Installing setuptools, pip, wheel...

  Error [Error 5] Access is denied while executing command C:\Users\augie\.platformio\penv\Scripts - setuptools pip wheel
...Installing setuptools, pip, wheel...done.

    at t.value (C:\Users\augie\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:25605)
    at <anonymous>
    at process._tickCallback (internal/process/next_tick.js:188:7)
```

pyr0ball, 2019-06-06:
+1 same issue for me

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets