platformio/platformio-atom-ide #2229: Python error compiling Marlin 2.0

biskero, 2019-07-12:
Hello,

I am compiling Marlin 2.0 with platformio and getting the following error related to Python I think, any suggestions?
I tried to re-install Atom and Platformio but still getting this error:
thx
```
Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip; board: nxp_lpc1768; frame
work: arduino)
--------------------------------------------------------------------------------
PlatformManager: Installing master
Error: Please read http://bit.ly/package-manager-ioerror
Error: Traceback (most recent call last):
  File "/Users/biskero/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 88, in main

    cli(None, None, None)
  File "/Users/biskero/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/Users/biskero/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/Users/biskero/.platformio/penv/lib/python2.7/site-packages/platformio/commands/__init__.py", line 37, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/Users/biskero/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/biskero/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/biskero/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/Users/biskero/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/biskero/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run/command.py", line 112, in cli
    result = (envname, ep.process())
  File "/Users/biskero/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run/processor.py", line 62, in proces
s
result = self._run_platform()
  File "/Users/biskero/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run/processor.py", line 113, in _run_
platform
    skip_default_package=True)
  File "/Users/biskero/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 555, in invoke

    return callback(*args, **kwargs)
  File "/Users/biskero/.platformio/penv/lib/python2.7/site-packages/platformio/commands/platform.py", line 302, in platform_i
nstall
    force=force):
  File "/Users/biskero/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 82, in install
    force=force)
  File "/Users/biskero/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 724, in install
    track=True)
  File "/Users/biskero/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 504, in _install_fr
om_url
    dlpath = self.download(url, tmp_dir, sha1)
  File "/Users/biskero/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 194, in download
    raise e
SSLError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /p3p/pio-nxplpc-arduino-lpc176x/ar
chive/master.zip (Caused by SSLError(SSLError(1, u'[SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:5
90)'),))
```

ivankravets, 2019-07-23:
SSLError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /p3p/pio-nxplpc-arduino-lpc176x/ar
chive/master.zip (Caused by SSLError(SSLError(1, u'[SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:5
90)'),))

It looks like Python installation is broken or you use old version of Python

biskero, 2019-07-23:
I have Python 2.7 installed on macOS, which one should I have?
thx

ivankravets, 2019-07-23:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

biskero, 2019-07-23:
echo $PATH
/opt/local/bin:/opt/local/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin
echo $PYTHONPATH
return nothing
pip not found

I guess I need to set the PYTHONPATH and install pip ?

ivankravets, 2019-07-23:
No, you do not need to set up PYTHONPATH. What is your macOS version?

biskero, 2019-07-23:
macOS 10.12.6 Sierra

wwwiesel, 2019-09-05:
Got the same issue here, using MacOS Mojave
Any hints how to fix that?

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets