platformio/platformio-atom-ide #2233: Platformio-ide installer hanging on Atom Dependencies after Atom 1.39 update

NewcDukem, 2019-07-24:
It seems that issue was fixed. Please reopen if you still experience it.

_Originally posted by @ivankravets in https://github.com/platformio/platformio-atom-ide/issues/2059#issuecomment-443012151_

Having issues downloading the platformio-ide. Hangs forever on "Atom Dependencies" (see referenced issue). Tried uninstalling Atom and reinstalling, but it immediately attempts to install the package again. Thus I am unable to use Atom at all. This happened after updating to Atom 1.39.
I did also try using 'apm install build' to fix the issue as described in the referenced issue, but no luck. Help!

NiOsY, 2019-07-24:
I too have the same problem during a clean install.
I tried uninstalling it based on the [Link](http://docs.platformio.org/en/latest/ide/atom.html#uninstall-atom-with-platformio-ide) but it didn't help.

NewcDukem, 2019-07-24:
I've regained use of Atom thankfully by uninstalling and deleting the **~\.atom** folder under users. Attempted to reinstall _platformio-ide-terminal_ and _platformio-ide_ but neither will install at all now. Luckily it doesn't freeze during the _Atom Dependencies_ step anymore which is nice.

MarinVukosav, 2019-07-25:
Same issue here.
`apm install build` also gets stuck and does nothing

humcarl, 2019-07-25:
temngo el mismo problema de newduckem, ayuda

Joshuas193, 2019-07-26:
I am having the same problem as those above. It gets stuck while installing on the Atom dependencies step of the install.

dadyday, 2019-07-26:
seems there is a problem with PlatformIO IDE Terminal.
edit ~/.atom/packages/platformio-ide/lib/config.js and comment out
```
export const ATOM_DEPENDENCIES = {
...
  /*'platformio-ide-terminal': {
    requirements: '>=2.5.0'
  },*/
```

MarinVukosav, 2019-07-26:
`apm install busy-signal` has solved the problem for me

iicky, 2019-07-26:
Same issue here after upgrading to Atom 1.39.1. I have tried uninstalling Platformio and reinstalling.

[image]

NewcDukem, 2019-07-26:
> apm install busy-signal has solved the problem for me

Did you do this before attempting to install platformio-ide, or during when it's frozen in installation?

MarinVukosav, 2019-07-26:
I uninstalled everything, then installed Atom 1.38.2
and then tried to install Platformio, but it got stuck on Dependencies again.
In the background there was an Atom notification, blue square pop-up in the top right corner that said something like "busy-signal dependency not installed ..." 
and then with that command in cmd it worked (while the frozen installation was going on)

humcarl, 2019-07-26:
la solucion es ir a usarios y eliminar todos los archivos de atom y luego
volver a instalar atom

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus. www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

NewcDukem, 2019-07-26:
This doesn't solve the original issue though. _Platformio-ide_ and _terminal_ do not work with Atom 1.39.0.

humcarl, 2019-07-26:
eso si es cierto

NewcDukem, 2019-07-26:
Just tried to install _platformio-ide_ again, then while installation hangs, input `apm install build` and `apm install busy-signal` but still hangs on Atom Dependencies. Hmm...

NewcDukem, 2019-07-26:
In the config.js file in the platformio there's a line `export const ATOM_DEPENDENCIES =...` I just copied every dependency listed there and installed with `apm install`. They all worked except for `build`. The list I got was: 
`'busy-signal' 'platformio-ide-terminal' 'platformio-ide-debugger' 'autocomplete-clang' 'linter-ui-default' 'intentions' 'linter' 'linter-gcc' 'language-ini' 'tool-bar' 'file-icons' 'build-platformio' 'build'`

_Originally posted by @stibinator in https://github.com/platformio/platformio-atom-ide/issues/2059#issuecomment-414883209_

I manually installed everything on @stibinator 's list. The only package that gave me a problem was `apm install platformio-ide-terminal`. Whereas everything else completed install.

Overall, the platformio-ide install is still hanging on Atom Dependencies.

NiOsY, 2019-07-26:
I see in platformio-atom-ide readme.md that "build failing". What does this mean?
I use Windows version of Atom, how can I install of platformio-ide early version for example 2.4.0 ?
I noticed any version of Atom always installs the latest platformio-ide.

iicky, 2019-07-26:
I ran into this issue on macOS Mojave when upgrading to Atom 1.39.

[image]

I uninstalled PlatformIO and tried to reinstall it again, but no luck. I disabled the `platformio-ide-terminal` dependency per @dadyday's [solution](https://github.com/platformio/platformio-atom-ide/issues/2233#issuecomment-515377184), and everything worked except for the terminal.

I tried reinstalling the `platformio-ide-terminal` package using `apm install platformio-ide-terminal`, but ran into the following error:

```
> node-pty-prebuilt@0.7.6 install /private/var/folders/rx/_k3qrh3d1xv9_jh_ynz9ntsr0000gn/T/apm-install-dir-119626-12485-tvn0hd.5omj/node_modules/platformio-ide-terminal/node_modules/node-pty-prebuilt
> prebuild-install || node scripts/install.js

prebuild-install WARN install No prebuilt binaries found (target=3.1.10 runtime=electron arch=x64 libc= platform=darwin)
gyp ERR! configure error 
gyp ERR! stack Error: Python executable "/Applications/Atom.app/Contents/Resources/app/apm/bin/python-interceptor.sh" is v3.7.3, which is not supported by gyp.
gyp ERR! stack You can pass the --python switch to point to Python >= v2.5.0 & < 3.0.0.
gyp ERR! stack     at PythonFinder.failPythonVersion (/Applications/Atom.app/Contents/Resources/app/apm/node_modules/npm/node_modules/node-gyp/lib/configure.js:501:19)
gyp ERR! stack     at PythonFinder.<anonymous> (/Applications/Atom.app/Contents/Resources/app/apm/node_modules/npm/node_modules/node-gyp/lib/configure.js:483:14)
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:267:7)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at maybeClose (internal/child_process.js:925:16)
gyp ERR! stack     at Socket.stream.socket.on (internal/child_process.js:346:11)
gyp ERR! stack     at emitOne (events.js:116:13)
gyp ERR! stack     at Socket.emit (events.js:211:7)
gyp ERR! stack     at Pipe._handle.close [as _onclose] (net.js:554:12)
gyp ERR! System Darwin 18.6.0
gyp ERR! command "/Applications/Atom.app/Contents/Resources/app/apm/bin/node" "/Applications/Atom.app/Contents/Resources/app/apm/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild" "--build_v8_with_gn=false"
gyp ERR! cwd /private/var/folders/rx/_k3qrh3d1xv9_jh_ynz9ntsr0000gn/T/apm-install-dir-119626-12485-tvn0hd.5omj/node_modules/platformio-ide-terminal/node_modules/node-pty-prebuilt
gyp ERR! node -v v8.9.3
gyp ERR! node-gyp -v v3.7.0
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/private/var/folders/rx/_k3qrh3d1xv9_jh_ynz9ntsr0000gn/T/apm-install-dir-119626-12485-tvn0hd.5omj/package.json'
npm WARN apm-install-dir-119626-12485-tvn0hd.5omj No description
npm WARN apm-install-dir-119626-12485-tvn0hd.5omj No repository field.
npm WARN apm-install-dir-119626-12485-tvn0hd.5omj No README data
npm WARN apm-install-dir-119626-12485-tvn0hd.5omj No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-pty-prebuilt@0.7.6 install: `prebuild-install || node scripts/install.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-pty-prebuilt@0.7.6 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```

I have no idea if this is related to the Atom upgrade, but following the [solution here](https://discuss.atom.io/t/npm-and-gyp-errors-during-apm-install/54658/7) worked for me. Looks like `gyp` was using Python 3 instead of Python 2. I have both Python 3 and Python 2.7 installed with Homebrew and the command `npm config set python python2.7` made it so I could install `platformio-ide-terminal` again.

NewcDukem, 2019-07-26:
Hmm, I'm indisposed for a day or two now. Anyone else with Windows wanna give iicky's solution a try?

NiOsY, 2019-07-26:
My solution are:
- install python manually (with add to path)
- restart windows
- install version 1.38.2 of Atom    because the versions above contain some major changes
- install platformio-ide
 Done :)
------------------my other problem--------
 platformio made several errors while compiling "...\firmware.elf section .text' will not fit in region iram1_0_seg'
- the solution is the lower version of platformio
`pip uninstall platformio`
`pip install platformio==3.6.7`

PGlyre, 2019-07-29:
I have the same problem with it hanging out and I curse myself for forgetting to turn off "Automatically Update" ......... the same problem about the time last year when Atom updated.

lgritsch, 2019-07-29:
Same Problem here...

samach123, 2019-07-29:
I noticed the line `gyp ERR! node -v v8.9.3` and sobefore going into Atom to install Platformio  I updated my node to v12.17.0 and when I tried `apm install platformio-ide-terminal` it worked. (I had already previously `apm` installed the other packages as listed by @NewcDukem.

NewcDukem, 2019-07-30:
> I noticed the line gyp ERR! node -v v8.9.3 and sobefore going into Atom to install Platformio  I updated my node to v12.17.0 and when I tried apm install platformio-ide-terminal it worked. (I had already previously apm installed the other packages as listed by @NewcDukem.

This worked with Atom 1.39.0/1 or 1.38.2?

NewcDukem, 2019-07-30:
I was able to get everything working with 1.38.2 again, but not 1.39.0 or 1.39.1

samach123, 2019-07-31:
> This worked with Atom 1.39.0/1 or 1.38.2?

Atom v1.39.1

ivankravets, 2019-07-31:
Depends on https://github.com/daviwil/node-pty-prebuilt/issues/21

/CC @daviwil

rlcayaco, 2019-07-31:
Please help. I dont know what happened. I just woke up and suddenly I saw a rebuild failed then I tried reinstalling platformio-ide-terminal and this error appeared.

_Installing “platformio-ide-terminal@2.9.1” failed.Hide output…

> node-pty-prebuilt@0.7.6 install C:\Users\PME\AppData\Local\Temp\apm-install-dir-119631-11424-3dcqqe.n5dvs\node_modules\platformio-ide-terminal\node_modules\node-pty-prebuilt
> prebuild-install || node scripts/install.js

C:\Users\PME\AppData\Local\Temp\apm-install-dir-119631-11424-3dcqqe.n5dvs\node_modules\platformio-ide-terminal\node_modules\node-pty-prebuilt>if not defined npm_config_node_gyp (node "C:\Users\PME\AppData\Local\atom\app-1.39.1\resources\app\apm\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild --build_v8_with_gn=false )  else (node "C:\Users\PME\AppData\Local\atom\app-1.39.1\resources\app\apm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild --build_v8_with_gn=false ) 

prebuild-install WARN install No prebuilt binaries found (target=3.1.10 runtime=electron arch=x64 libc= platform=win32)
gyp ERR! configure error 
gyp ERR! stack Error: Can't find Python executable "C:\ProgramData\Anaconda3\python.EXE", you can set the PYTHON env variable.
gyp ERR! stack     at PythonFinder.failNoPython (C:\Users\PME\AppData\Local\atom\app-1.39.1\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\configure.js:492:19)
gyp ERR! stack     at PythonFinder.<anonymous> (C:\Users\PME\AppData\Local\atom\app-1.39.1\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\configure.js:517:16)
gyp ERR! stack     at C:\Users\PME\AppData\Local\atom\app-1.39.1\resources\app\apm\node_modules\npm\node_modules\graceful-fs\polyfills.js:284:29
gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:152:21)
gyp ERR! System Windows_NT 10.0.17134
gyp ERR! command "C:\\Users\\PME\\AppData\\Local\\atom\\app-1.39.1\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\PME\\AppData\\Local\\atom\\app-1.39.1\\resources\\app\\apm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild" "--build_v8_with_gn=false"
gyp ERR! cwd C:\Users\PME\AppData\Local\Temp\apm-install-dir-119631-11424-3dcqqe.n5dvs\node_modules\platformio-ide-terminal\node_modules\node-pty-prebuilt
gyp ERR! node -v v8.9.3
gyp ERR! node-gyp -v v3.7.0
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\PME\AppData\Local\Temp\apm-install-dir-119631-11424-3dcqqe.n5dvs\package.json'
npm WARN apm-install-dir-119631-11424-3dcqqe.n5dvs No description
npm WARN apm-install-dir-119631-11424-3dcqqe.n5dvs No repository field.
npm WARN apm-install-dir-119631-11424-3dcqqe.n5dvs No README data
npm WARN apm-install-dir-119631-11424-3dcqqe.n5dvs No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-pty-prebuilt@0.7.6 install: `prebuild-install || node scripts/install.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-pty-prebuilt@0.7.6 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\PME\.atom\.apm\_logs\2019-07-31T08_42_52_805Z-debug.log_

benhadad, 2019-08-01:
Sounds like https://github.com/daviwil/node-pty-prebuilt hasn't had time to update the prebuilt files to be compatible with the latest Atom.  I am new and just completed getting my build platform up and running and had my first project due this week. I had not reboot or restarted my atom setups until today, wish I too had turned off automatic updates :(.  Well team not impress by the Atom/PlatformIO setup now. Guess back to VS....

rlcayaco, 2019-08-01:
They say Atom version 1.38.2 is working fine. Have you tried that?

Erokawa, 2019-08-01:
It happend me the same issue with the last version of atom. I installed this time the 1.38.1 version and it worked

rlcayaco, 2019-08-01:
I dont know what happened but Atom just wont open. I tried reinstalling it
now it wont let me install its giving me errors.

the-j0k3r, 2019-08-01:
See https://github.com/platformio/platformio-atom-ide-terminal/pull/753 for solution awaiting @ivankravets merge and release

rlcayaco, 2019-08-02:
I guess I will just wait for it to get updated and fixed.

jorgedepaz, 2020-04-04:
> seems there is a problem with PlatformIO IDE Terminal.
> edit ~/.atom/packages/platformio-ide/lib/config.js and comment out
> 
> ```
> export const ATOM_DEPENDENCIES = {
> ...
>   /*'platformio-ide-terminal': {
>     requirements: '>=2.5.0'
>   },*/
> ```

This was useful for me, thanks

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets

hjalmer, 2020-11-12:
There are still problems installing "PlatformIO IDE" on macOS Catalina v10.15.7.
I tried to uninstall and install again however the "Atom dependencies" is still hanging... i left i for a few hours with no result.

Any ideas how to fix this?

Krzysztof-acopotrzeba, 2020-11-14:
Hi, I had the same problem. I've deactivated my antivirus set and reinstalled Atom and it's working now.

willdmo, 2020-11-27:
I tried all the solutions here. Nothing worked. Atom still hanged and all tries to install platform-ide-terminal manually as well. I switched to vscode, to be able to work

abdelrahmandakak, 2020-12-05:
I have the same problem. I guess I'm going to switch to VSCODE