platformio/platformio-atom-ide #2247: platfromio-ide crashes atom

Ofloo, 2019-08-11:
When I open atom, it once it loads platformio-ide it crashes, keeps asking to reload but never starts, .. when I then remove platformio-ide from the atom package directory, it starts fine .. when I reinstall it works but if i close the window and reopen it, it does the same thing all over again.

dave2492, 2019-10-04:
I've had the same experience on two different linux machines (one was a chromebook running crouton - so I tried a linux VM running Ubuntu Mate 18.04 and got the same results).  In both cases I installed Atom via the Ubuntu PPA.  Initially it works fine including compiling the Marlin firmware, but crashes when I exit and restart Atom and it automatically re-opens the project.  If I move the project directory prior to starting Atom it doesn't crash.  I've also noticed lots (like nearly 400) zombie processes are created when Atom begins to load leading up to the crash.

Atom version 1.40.1 x64
PlatformIO says: Home 2.3.3, Core 4.0.3

dave2492, 2019-10-04:
I noticed a dump file in /tmp so here it is.  I also tried removing the build products at .pio/build/LPC1768, and then restoring the project dir to where Atom was looking for it.  Several restarts and no crash.  However after I built again and restarted, I still don't see a crash.  Oh well.
But here is the dump...
[atomdump.zip](https://github.com/platformio/platformio-atom-ide/files/3692053/atomdump.zip)

Ofloo, 2019-10-05:
I've noticed when I remove .config/Atom then it works just fine. Just once :p

Chuckonelli, 2019-10-24:
I have the same issue in Linux Mint 19.2. 
What is exactly the solution? I'm quite new to Linux and tried solving this issue for 2-3 days without any luck.

dave2492, 2019-10-24:
I have been having success (usually) by removing the final build results
before starting Atom.  For me, that is a "rm
src/Marlin/.pio/build/LPC1768/firmware.*"  Good luck!
I can't explain why the platformio folks aren't interested in fixing a
crash...

Chuckonelli, 2019-10-24:
Thanks for your answer. I just switched to Visual Studio Code. I noticed right away the quicker startup and performance overall. PlatformIO works just fine. 
How its done:
https://platformio.org/install/ide?install=vscode
It may be the best fix ;)

dlarue, 2019-10-30:
When I removed the one offending project(sonoff Tasmota) I was able to load a few Arduino-Mini projects and compile fine. Then I tried to do an update and it was complaining platformio-ide-terminal package wasn't loaded.  I clicked a red bug icon at the bottom of the window and it showed a package was out of data and gave an option to rebuild. I rebuilt the terminal package and was able to open the Tasmota project without atom crashing.

I'm having other problems with the Tasmota platformIO build on Ubuntu 18.04 but that's another issue.

SynAckFin, 2020-01-20:
Same issue here running it on OpenSUSE-15.0.
rpm: atom-1.32.2-0.1.x86_64
Version:
```
Atom    : 1.32.2
Electron: 2.0.9
Chrome  : 61.0.3163.100
Node    : 8.9.3

```

When I run it from a terminal with the -f flag I get the following output:
```
[1854:0100/000000.090037:FATAL:memory.cc(22)] Out of memory. size=16384
ExceptionHandler::GenerateDump sys_pipe failed:Too many open files
ExceptionHandler::SendContinueSignalToChild sys_write failed:Bad file descriptorExceptionHandler::WaitForContinueSignal sys_read failed:
Bad file descriptor
[1854:0100/000000.092686:FATAL:memory.cc(22)] Out of memory. size=262144
Failed to generate minidump.Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
```
If I run it in development mode (-d) then it starts up without any issues although the start up seems a little slower.

fractalfederation, 2020-02-03:
im having the same problem on ubuntu 18.04 and 19.10
i tried deleting .plaformio and running pio upgrade --dev
no dice

dlarue, 2020-02-03:
Try looking for any hanging processes( ps -fe ) for platformio and kill them before restarting the IDE.

I've not run PlatformIO in a while, saw your post and opened it and saw the AVR and ESP parts needed updating and after updating I could not open a project. Tried 3 times and got a hung IDE until I saw there was a hanging process. After killing it and restarting PlatformIO I could load the Tasmota project and compile it.

fractalfederation, 2020-02-12:
i did find a process, but it didn't help. i uninstalled atom and platformio and reinstalled, hasnt happened since *shrugs*

Galaxie1964, 2020-04-15:
I'm having the same problem but i'v narrowed it down

to the Marlin platformio.ini file
to fix crashing on startup i turned off restore previous windows on start
i got around the crash by opening multiple instances the first one crashes 
next i'v replaced the Marlin .ini with a fresh one from a new project, the crashing has stopped and the project is reliably opening now.

now i'm not sure what exactly about the Marlin file is causing the problem. but it seems to be the problem

the problem is crashing when loading Marlin 2.0 projects 
running
Ubuntu 18.04.4 LTS
Atom    : 1.45.0
Electron: 4.2.7
Chrome  : 69.0.3497.128
Node  : v10.11.0

Ofloo, 2020-04-17:
I don't know what happend, but today my theme has changed to one dark and up until now it hasn't crashed anymore.

Galaxie1964, 2020-04-18:
so this is what i found and explains allot
according to the marlin platformio page https://marlinfw.org/docs/basics/install_platformio.html
platformio is dropping support for atom or that's how it reads.
I switched to compiling in the terminal with platformio core and you can edit the files with whatever txt editor you like

dlarue, 2020-04-18:
It was very obvious long ago that PlatformIO did NOT want to support Atom and was heavily pushing Microsoft's editor.  HEAVILY.  Atom uses the Electron framework under-the-hood and so does Microsoft's editor and you know, developers, developers, developers.  One developer IDE to rule them all... With Microsoft owning github now, no doubt they plan to kill Atom. From what I read, there are lots of proprietary hooks(and plugins) in MS-Visual Studio Code(VSC). CLI!

Ofloo, 2020-04-19:
Still got the issue with platformio since the day I've made the ticket.

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets

dlarue, 2020-06-13:
I noticed long ago the bias against Atom and the aggressive push to MS VSCode. Not a fan of that kind of manipulation...

Ofloo, 2020-06-13:
even with the new update version 2.7.0 it still freezes when platformio-ide is installed ! Doing killall atom and then reloading it settles the install, then it says that:

PlatformIO IDE installation has been suspended.

Seems like PlatformIO IDE Installer is already started in another window.

So what it boils down to is that it still doesn't work properly.

dlarue, 2020-06-13:
They have been forcing an IDE change for a very long time so maybe it's time to try Eclipse.

TomaszKisiel, 2020-07-21:
Very sad that platformio doesn't suport atom on linux anymore :(