platformio/platformio-atom-ide #2260: Start PIO Home Server

DAMOCLESXXI, 2019-08-30:
Error: Error: Could not find a version that satisfies the requirement '~2.38.190418' for your system 'windows_amd64'

at C:\Users\Usuario\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:17225
at ChildProcess.h (C:\Users\Usuario\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:6033)
at ChildProcess.emit (events.js:182:13)
at ChildProcess.cp.emit (C:\Users\Usuario\.atom\packages\platformio-ide\node_modules\cross-spawn\lib\enoent.js:34:29)
at maybeClose (internal/child_process.js:961:16)
at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)

sphereinabox, 2019-10-15:
I hit the same issue with 64-bit vscode. Switching to 32-bit vscode fixed it for me.

luc-debug, 2019-10-15:
I have the same problem.
Unfortunately your suggestion does not solve it for me.
Do you have any other suggestions?

Cheeseborgers, 2019-10-15:
Same Issue here. was working fine yesterday.
Tried Python 3.6-3.8, both x86 and amd64 versions.

`CorePackageManager: Installing contrib-pysite @ ~2.38.190418
Error: Could not find a version that satisfies the requirement '~2.38.190418' for your system 'windows_amd64'`

And just "Loading..." with VS code.

Cheeseborgers, 2019-10-15:
Switching to X86 VS code did not work for me though.

swd120, 2019-10-16:
I'm also getting this error - both VS Code and Atom

yulong88888, 2019-10-27:
Guys, have your problems solved? I also encountered this problem. I have never encountered it before using PIO. Due to the reinstallation of the Windows system, I have problems. I think the reason is that contrib-pysite-windows_amd64-2.38.191019.tar cannot be downloaded.

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets