platformio/platformio-atom-ide #2261: pio device monitor "Error: Got unexpected extra arguments ()" or better: --echo '' vs --echo

cyberlussi, 2019-09-01:
Since updating to latest Atom.io and platformio packages I have an issue which I could not find in the issue list so far. It looks first like #62 but it is not matching.

Atom.io 1.40.1 for Mac
platformio-ide-terminal 2.9.4
platformio-ide 2.41 
PlatformIO Core 4.0.3

I just started via the toolbar the serial monitor as usual (with enabled local echo), but the opened terminal complains:

```
pio device monitor --port /dev/cu.usbserial-FT --baud 115200 --echo ''
Usage: pio device monitor [OPTIONS]

Error: Got unexpected extra argument ()
```

When calling ```pio device monitor --port /dev/cu.usbserial-FT --baud 115200 --echo``` it works as expected. So the command is not build correctly. The option "echo" from the terminal menu > advanced options > local echo > yes  is not correctly transformed to ```--echo``` but to ```--echo ''```

I nailed it down to https://github.com/platformio/platformio-atom-ide/blob/510125e60542858b5d92db5efb72ba2ab37a988e/lib/serial-monitor/command.js#L43-L47

Here (e.g. L45), I think, a check should be added whether the ```${settings[key]}``` is not empty to implement the valueless --option correctly. Or, the way the options are handled could be optimized.

I am not sure why this happens now, but maybe the terminal command line options changed recently ... looking on the options of the terminal, --raw and others could be also affected, although not all are used in the UI menu for the terminal.

```
Usage: pio device monitor [OPTIONS]

Options:
  -p, --port TEXT              Port, a number or a device name
  -b, --baud INTEGER           Set baud rate, default=9600
  --parity [N|E|O|S|M]         Set parity, default=N
  --rtscts                     Enable RTS/CTS flow control, default=Off
  --xonxoff                    Enable software flow control, default=Off
  --rts INTEGER RANGE          Set initial RTS line state
  --dtr INTEGER RANGE          Set initial DTR line state
  --echo                       Enable local echo, default=Off
  --encoding TEXT              Set the encoding for the serial port (e.g.
                               hexlify, Latin1, UTF-8), default: UTF-8
  -f, --filter TEXT            Add text transformation

  --eol [CR|LF|CRLF]           End of line mode, default=CRLF
  --raw                        Do not apply any encodings/transformations
  --exit-char INTEGER          ASCII code of special character that is used to
                               exit the application, default=3 (Ctrl+C)
  --menu-char INTEGER          ASCII code of special character that is used to
                               control miniterm (menu), default=20 (DEC)
  --quiet                      Diagnostics: suppress non-error messages,
                               default=Off
  -d, --project-dir DIRECTORY
  -e, --environment TEXT       Load configuration from `platformio.ini` and
                               specified environment
  -h, --help                   Show this message and exit.
```

cyberlussi, 2020-01-04:
Any update on this?

GewoonGijs, 2020-01-07:
I experienced the same issue. And indeed for the option --raw the behaviour fails in a similar way.

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets

sgofferj, 2021-05-31:
The issue is still present in 2.7.2. Please reopen.