platformio/platformio-atom-ide #2265: Start PIO Home Server

nealre, 2019-09-03:
Tried to install Atom and Platformio.ide and Got an error, 64 bit AMD issue?

Error: Error: Could not find a version that satisfies the requirement '~2.38.190418' for your system 'windows_amd64'

at C:\Users\Owner\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:17372
at ChildProcess.h (C:\Users\Owner\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:6150)
at ChildProcess.emit (events.js:182:13)
at ChildProcess.cp.emit (C:\Users\Owner\.atom\packages\platformio-ide\node_modules\cross-spawn\lib\enoent.js:34:29)
at maybeClose (internal/child_process.js:961:16)
at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)

Thanks for Your Help
Rick

sphereinabox, 2019-10-15:
This looks like a dupe of #2260

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets