platformio/platformio-atom-ide #2280: Error: Detected unknown package 'framework-arduino-lpc176x'

deckerchevy, 2019-10-03:
I am trying to build the firmware for skr1.3 and followed the guide from Teaching tech. https://www.youtube.com/watch?v=GNGN2iSQ5j4 exactly. I try to build and it gives me the below error everytime. I don't know how to fix it.

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip; framework: arduino; board: nxp_lpc1768)
-------------------------------------------------------------------------------
PackageManager: Installing framework-arduino-lpc176x @ >=0.0.2
Error: Detected unknown package 'framework-arduino-lpc176x'

deckerchevy, 2019-10-03:
I also get this error when trying to do the same procedure via Visual Studio Code. Just an FYI.

CUI233333, 2020-04-17:
I also get this error when trying to do the same procedure via Visual Studio Code when building Marlin

CUI233333, 2020-04-18:
i solved it.just the internet connection。try your phone connection.

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets