platformio/platformio-atom-ide #2287: Start PIO Home Server - home not loading, not build

Psh777, 2019-10-19:
Error: Detected unknown package 'contrib-pysite'

at /Users/psh/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:17383
at ChildProcess.h (/Users/psh/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:6150)
at ChildProcess.emit (events.js:182:13)
at maybeClose (internal/child_process.js:961:16)
at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)

Psh777, 2019-10-19:
atom 1.40.1 x64
platformio ide 2.4.1
macos

ronzeiller, 2019-10-26:
Same here:
atom 1.40.1 x64
platformio ide 2.4.1 and same on "use development version of PlatformIO IDE"
PlatformIO, version 4.0.3
macos Sierra

Error: Detected unknown package 'contrib-pysite'
at /Users/ronnie/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:17820
at ChildProcess.h (/Users/ronnie/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:6292)
at ChildProcess.emit (events.js:182:13)
at maybeClose (internal/child_process.js:961:16)
at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)

ronzeiller, 2019-10-28:
Got it working somehow...

Deleted directory .platformio in /User/xy
Copied .platformio (old version) from an old backup
Started Atom again

And now The "Home"-Button deliveres the known "Welcome to PlatformIO" again

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets