platformio/platformio-atom-ide #2304: Start PIO Home Server

Pirelly1970, 2019-10-27:


Pirelly1970, 2019-10-27:
I have a problem if i open atom i can not open the platform io
[image]

dlukacic, 2019-10-29:
I have the same problem.

Pirelly1970, 2019-10-31:
Ok..but what will we do now...is here anybody they will help us
I install atom on a parralels windows platform but there. The install version is broken...HELP please

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets