platformio/platformio-atom-ide #2313: Start PIO Home Server

kieferdev, 2019-11-04:
I have a problem while starting platformio

at C:\Users\Sebastian\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:10359
at ChildProcess.h (C:\Users\Sebastian\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:6292)
at ChildProcess.emit (events.js:182:13)
at ChildProcess.cp.emit (C:\Users\Sebastian\.atom\packages\platformio-ide\node_modules\cross-spawn\lib\enoent.js:34:29)
at maybeClose (internal/child_process.js:962:16)
at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)

jlindeh, 2019-11-25:
Hello,

I have the same problem, have you solved it?

kieferdev, 2019-11-25:
Hello, 
I solved it by make an update for atom and platformIO

closed 2019-11-25 by kieferdev

jlindeh, 2019-11-25:
Hello  Sebastian,

Ok thanks, I’ll try.