platformio/platformio-atom-ide #2322: InstallationManager

ebear01, 2019-11-09:


ebear01, 2019-11-09:
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv Error: Virtualenv Create: Traceback (most recent call last): File "C:\Users\beerh.platformio.cache\tmp-45468zZ4ur6yHOk6h\virtualenv-16.7.5\virtualenv.py", line 2632, in main() File "C:\Users\beerh.platformio.cache\tmp-45468zZ4ur6yHOk6h\virtualenv-16.7.5\virtualenv.py", line 870, in main symlink=options.symlink, File "C:\Users\beerh.platformio.cache\tmp-45468zZ4ur6yHOk6h\virtualenv-16.7.5\virtualenv.py", line 1156, in create_environment install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink) File "C:\Users\beerh.platformio.cache\tmp-45468zZ4ur6yHOk6h\virtualenv-16.7.5\virtualenv.py", line 1526, in install_python shutil.copyfile(executable, py_executable) File "C:\Users\beerh.platformio\python37\lib\shutil.py", line 121, in copyfile with open(dst, 'wb') as fdst: PermissionError: [Errno 13] Permission denied: 'C:\Users\beerh\.platformio\penv\Scripts\python.exe'

Using base prefix 'C:\Users\beerh\.platformio\python37' New python executable in C:\Users\beerh.platformio\penv\Scripts\python.exe

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets