platformio/platformio-atom-ide #2325: InstallationManager

ghclifton, 2019-11-11:
installed Atom for use in installing firmware but received following error when installing platformio-ide..
[image]

any help appreciated.

thanks
Gary

dant220, 2019-11-19:
Try this:
sudo apt-get install python3-distutils
sudo apt-get install python3-venv

But I see you are using windows. In any case you need to install distutils and venv. For your version of python.

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets