platformio/platformio-atom-ide #2336: pio "stealing" python environment

cmburg, 2019-11-24:
I am using Atom for nearly everything (LaTeX, Arduino stuff, and now python).
However, since pio is advertised as a hardware development tool, it is messing around with the python environments.

Running the code 
`import sys`
`print(sys.executable)`
within Atom I get
`/home/myname/.platformio/penv/bin/python
`.

But when I run the same file from terminal I get
`/usr/bin/python`

This behaviour is problematic (at least for me) since I am installing python packages globally wich cannot be found within Atom. Installing all packages within Atom might be a solution but I am running my OS's terminal besides Atom since I am unable to display Atoms terminal in a split left-right fashion.

So how do I alter Atom to use the global python-"stuff"?

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

oxihormesis, 2020-10-02:
I'm on the latest release. I use pip3 to install Django in some virtualenv. I then try a Django command and it says django module not found. I try to install Django again to fix this - it says `Requirement already satisfied: Django in /home/user/.platformio/penv/lib/python3.6/site-packages (3.1.2)`