platformio/platformio-atom-ide #2349: InstallationManager / in ATOM / on Mac

cmmoellering, 2019-12-22:
When trying to install PlatformIO in ATOM I get the following error inside the Installationmanager:

Error: PIP Core: /opt/local/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python: No module named pip

at s.runCommand (/Users/christian.moellering/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:35177)
at ChildProcess.d (/Users/christian.moellering/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:8308)
at ChildProcess.emit (events.js:182:13)
at maybeClose (internal/child_process.js:962:16)
at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)

Strange enough, on my system is in none of the finely working python installations a module named pip..? And I can do installations of python packages with pip or pip3 respectively without problem. Where or how does ATOM searches for PIP?

ivankravets, 2019-12-23:
Could you try to restart Atom and try again?

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets