platformio/platformio-atom-ide #2373: InstallationManager

jelacio, 2020-03-27:
I have been with this problem from the start sorry but I'm ignorant and I try finding the help online but I can't find it  
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv Error: Can not find Python Interpreter. Please install Python 3.5 or above manually at t.value (C:\Users\jlh16.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:34170) at process._tickCallback (internal/process/next_tick.js:68:7)

ivankravets, 2020-06-13:
Please upgrade to v2.7.0 where the new installer is added https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets