platformio/platformio-atom-ide #2375: InstallationManager

deutrion, 2020-03-29:
/*
STM32 adaption by Matthias Diro, tested with maple mini and heltec OLED 12864 I2c; adress: 0x3C (SPI should work, but I didn't own one)
Things to know:
 This adaption uses hardware I2C (now Wire.h), Port: I2c2. SDA=0, SCL=1 on maple mini
 further details: STM32_README.txt
*/
/*********************************************************************
This is an example for our Monochrome OLEDs based on SSD1306 drivers

  Pick one up today in the adafruit shop!
  ------> http://www.adafruit.com/category/63_98

This example is for a 128x64 size display using I2C to communicate
3 pins are required to interface (2 I2C and one reset)

Adafruit invests time and resources providing this open source code, 
please support Adafruit and open-source hardware by purchasing 
products from Adafruit!

Written by Limor Fried/Ladyada  for Adafruit Industries.  
BSD license, check license.txt for more information
All text above, and the splash screen must be included in any redistribution
*********************************************************************/

#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306_STM32.h>

#define OLED_RESET 4
Adafruit_SSD1306 display(OLED_RESET);

#define NUMFLAKES 10
#define XPOS 0
#define YPOS 1
#define DELTAY 2

#define LOGO16_GLCD_HEIGHT 16 
#define LOGO16_GLCD_WIDTH  16 
static const unsigned char PROGMEM logo16_glcd_bmp[] =
{ B00000000, B11000000,
  B00000001, B11000000,
  B00000001, B11000000,
  B00000011, B11100000,
  B11110011, B11100000,
  B11111110, B11111000,
  B01111110, B11111111,
  B00110011, B10011111,
  B00011111, B11111100,
  B00001101, B01110000,
  B00011011, B10100000,
  B00111111, B11100000,
  B00111111, B11110000,
  B01111100, B11110000,
  B01110000, B01110000,
  B00000000, B00110000 };

#if (SSD1306_LCDHEIGHT != 64)
#error("Height incorrect, please fix Adafruit_SSD1306.h!");
#endif

byte Puerto = 0;
unsigned long previousMillis = 0;
const unsigned long interval = 200;

void setup()   {  
  pinMode(LED_BUILTIN, OUTPUT);              
  Serial.begin(9600);

  // by default, we'll generate the high voltage from the 3.3v line internally! (neat!)
  display.begin(SSD1306_SWITCHCAPVCC, 0x3C);  // initialize with the I2C addr 0x3D (for the 128x64)
  // init done
  
  display.display();  // Despliega el contenido de la memmoria intrna del display
  delay(200);

  // Clear the buffer.
  display.clearDisplay();
  display.display();
}

void loop() {
   unsigned long currentMillis = millis();
  if (currentMillis - previousMillis > interval){
    previousMillis = currentMillis;
    Serial.println("millis" + String(currentMillis));
  digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
    
  }
 for(Puerto=0;Puerto<255;Puerto++){
  delay(20);
  EscribeBit();
  
 }
  
}

void EscribeBit(void){
  display.setTextColor(WHITE,BLACK);
  byte x1 = 0;
  byte x2 = 0;
  display.setTextSize(1);
  for(int i=0;i<8;i++){
    x1 = bitRead(Puerto,i);
    x2=8-i;
    
    display.setCursor(x2*12,0);
    if(x1==0){
      display.write(95);
    }
    else{
      display.write(42);
    }
   // display.setCursor(0,0);
   // display.println("BMP no encontrado, verifique el circuito");
    
  }
  display.display();
}

ivankravets, 2020-06-13:
Please upgrade to v2.7.0 where the new installer is added https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets