platformio/platformio-atom-ide #2377: LUBUNTU Atom: Failed to activate the platformio-ide package

landalv, 2020-04-05:
Error: Cannot find module 'platformio-node-helpers'
    at Module._resolveFilename (module.js:470:15)
    at Module._resolveFilename (/opt/atom/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.get_Module._resolveFilename (/opt/atom/resources/app/src/module-cache.js:354:58)
    at Module.require (file:///opt/atom/resources/app.asar/static/index.js:40:43)
    at require (/opt/atom/resources/app/src/native-compile-cache.js:66:33)
    at Object.<anonymous> (/home/carlos/.atom/packages/platformio-ide/lib/maintenance.js:24:30)
    at Object.<anonymous> (/home/carlos/.atom/packages/platformio-ide/lib/maintenance.js:154:3)
    at Module.get_Module._compile (/opt/atom/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/opt/atom/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///opt/atom/resources/app.asar/static/index.js:47:45)
    at require (/opt/atom/resources/app/src/native-compile-cache.js:66:33)
    at Object.<anonymous> (/home/carlos/.atom/packages/platformio-ide/lib/main.js:29:20)
    at Object.<anonymous> (/home/carlos/.atom/packages/platformio-ide/lib/main.js:420:3)
    at Module.get_Module._compile (/opt/atom/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/opt/atom/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///opt/atom/resources/app.asar/static/index.js:47:45)
    at require (internal/module.js:20:19)
    at customRequire (/opt/atom/resources/app/static/<embedded>:96:26)
    at Package.requireMainModule (/opt/atom/resources/app/src/package.js:782:33)
    at Package.activateNow (/opt/atom/resources/app/src/package.js:220:40)
    at activationHookSubscriptions.add.packageManager.onDidTriggerActivationHook (/opt/atom/resources/app/src/package.js:901:81)
    at Function.module.exports.Emitter.simpleDispatch (/opt/atom/resources/app/node_modules/event-kit/lib/emitter.js:25:20)
    at Emitter.module.exports.Emitter.emit (/opt/atom/resources/app/node_modules/event-kit/lib/emitter.js:141:34)
    at PackageManager.triggerActivationHook (/opt/atom/resources/app/src/package-manager.js:732:40)
    at AtomEnvironment.updateProcessEnvAndTriggerHooks (/opt/atom/resources/app/src/atom-environment.js:980:25)
    at <anonymous>

neod123, 2020-04-15:
I have the same issue.
It came from a new install of atom

Hairybiker, 2020-05-30:
same issue, new install under Linux Mint Cinnamon 19.3 64bit

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets

waldoratti, 2020-07-02:
That means that we will not be able to use more Platformio in Atom ??? Currently perform an Atom installation, and it won't let me use Platformio, the Atom locks up and doesn't work

ivankravets, 2020-07-03:
Could you try to install firstly https://atom.io/packages/platformio-ide-terminal? You will need GCC.

svdb99, 2020-08-30:
Same issue with 2.7.0 on Atom/win10.
If you discontinued support for Atom why don't you officially say so instead of "highly" recommending to use something else. :(

zachlindblad, 2020-10-04:
Same issue with atom on windows 10, should remove Atom support from your site if it doesn't seem to support it anymore

eforneas, 2020-12-01:
Same issue MacBook macOs High Sierra ATOM 1.37.0 x64