platformio/platformio-atom-ide #2403: Debian Sid -- Atom -- platformio-ide 2.7.1 failed to activate

beckrd, 2020-06-14:
I feel like I have  been running in circles trying to sort this out.  Atom pointed me to issue 2377, which had a solution to use 2.7.0.  Is it really necessary to downgrade or is there another issue here?

`Unexpected token {
Hide Stack Trace
SyntaxError: Unexpected token {
    at Module.get_Module._compile (/usr/share/atom/resources/app/static/<embedded>:11:146461)
    at Object.value [as .js] (/usr/share/atom/resources/app/static/<embedded>:11:150231)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (file:///usr/share/atom/resources/app.asar/static/index.js:47:45)
    at require (/usr/share/atom/resources/app/static/<embedded>:11:145974)
    at Object.<anonymous> (/home/rdb/.atom/packages/platformio-ide/lib/maintenance.js:24:30)
    at Object.<anonymous> (/home/rdb/.atom/packages/platformio-ide/lib/maintenance.js:131:2)
    at Module.get_Module._compile (/usr/share/atom/resources/app/static/<embedded>:11:146684)
    at Object.value [as .js] (/usr/share/atom/resources/app/static/<embedded>:11:150231)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (file:///usr/share/atom/resources/app.asar/static/index.js:47:45)
    at require (/usr/share/atom/resources/app/static/<embedded>:11:145974)
    at Object.<anonymous> (/home/rdb/.atom/packages/platformio-ide/lib/main.js:11:19)
    at Object.<anonymous> (/home/rdb/.atom/packages/platformio-ide/lib/main.js:333:22)
    at Module.get_Module._compile (/usr/share/atom/resources/app/static/<embedded>:11:146684)
    at Object.value [as .js] (/usr/share/atom/resources/app/static/<embedded>:11:150231)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (file:///usr/share/atom/resources/app.asar/static/index.js:47:45)
    at require (internal/module.js:11:18)
    at customRequire (/usr/share/atom/resources/app/static/<embedded>:1:637521)
    at Package.requireMainModule (/usr/share/atom/resources/app/static/<embedded>:11:2924353)
    at Package.activateNow (/usr/share/atom/resources/app/static/<embedded>:11:2912487)
    at string.e.trim.length.activationHookSubscriptions.add.packageManager.onDidTriggerActivationHook (/usr/share/atom/resources/app/static/<embedded>:14:2445)
    at Function.simpleDispatch (/usr/share/atom/resources/app/static/<embedded>:11:1189948)
`

ivankravets, 2020-06-16:
1. Do you use the latest Atom version?
2. I'm not sure that this issue is related with our extension.

beckrd, 2020-06-16:
Many apologies.  I thought I had the latest version, but I was quite out of date.  Upgrading to the latest solved the issue I had.  Thanks.

closed 2020-06-16 by beckrd